kinx-project/kint #6: Bluetooth

SleepyBrett, 2020-08-01:
Whats the possibility of a slightly more complicated board (may require a multiplexer) with bluetooth support and lipo charging through the usb port? The adafruit feather has a nice through port lipo charging feature (though I think I did a pincount on the feather w/ bluetooth and found that a multiplexer would be required.

Other features that would be awesome, support of the built in LED windows, maybe with rgb leds.

As an aside, I'm super happy to see this rev to the old stapleberg board, I've got two modded advantage 1s and they make me very happy.

stapelberg, 2020-08-01:
> Whats the possibility of a slightly more complicated board (may require a multiplexer) with bluetooth support and lipo charging through the usb port? The adafruit feather has a nice through port [...]

You’re welcome to build such a board based on this one, but I personally have neither time nor motivation to do bluetooth for my keyboards :)

> Other features that would be awesome, support of the built in LED windows, maybe with rgb leds.

What do you mean? The caps lock/numlock/keypad/scroll lock LEDs already are supported

> As an aside, I'm super happy to see this rev to the old stapleberg board, I've got two modded advantage 1s and they make me very happy.

Very glad to hear! Enjoy :)

SleepyBrett, 2020-08-01:
gotcha on the LEDs I saw the pads on the old stapleberg board but could never quite got around to making them work.

I'll let you know if I make any progress on the bluetooth!

closed 2020-08-01 by SleepyBrett

stapelberg, 2020-08-02:
> gotcha on the LEDs I saw the pads on the old stapleberg board but could never quite got around to making them work.

Ah, alright! With the new revision, I know that they are working for sure, so let me know if you run into any trouble and we should be able to figure it out :)