kinx-project/kint #7: RESET quantum keycode does not work with Teensy 3.6

ameyp, 2020-08-08:
I've modded my Advantage 1 with a Teensy 3.6 and the RESET quantum keycode doesn't seem to work. When I run `qmk flash` and qmk is at `Waiting for Teensy device... (hint: press the reset button)`, sending the RESET keycode with a keypress doesn't put the board in bootloader mode and I have to press the physical button to allow flashing to proceed.

stapelberg, 2020-08-09:
Thanks for the report. I indeed never tested this, because I had just assumed that reset is only for restarting the keyboard, not for reprogramming it :) 

https://www.pjrc.com/teensy/jump_to_bootloader.html is relevant, but doesn’t contain an address for the Teensy 3.6.

Let me see if this might be an easy fix.

stapelberg, 2020-08-09:
Looks like on the Teensy 3.x, we need to call the BKPT instruction: https://forum.pjrc.com/threads/30521-Rebooting-Teensy-(LC-3-0)-to-bootloader-without-Teensyduino

stapelberg, 2020-08-09:
I think commit https://github.com/kinx-project/qmk_firmware/commit/bb8ab07b477b8de7175db24a04e1a3569028da98 might help, but can’t test it right now. Can you `git pull` in your branch, recompile, reflash, and try again?

stapelberg, 2020-08-09:
Just had a chance to test, and with the commit, the QK_RESET keycode jumps to the bootloader for me :)

closed 2020-08-09 by stapelberg

ameyp, 2020-08-09:
Works for me too, thanks!

lread, 2023-08-05:
Oh, this is cool, I can flash via my Teensy without having to press the button on the Teensy?

I think `RESET` might have since been renamed to `QK_BOOTLOADER` in [commit](https://github.com/qmk/qmk_firmware/commit/b45a037c7edcd9c1cfadceb69ed51e7d42aa58c1#diff-801585dcf530fa1c9c07e022ddd95783097bfda2f180da6889ba64d6d6e3042aR224).

lread, 2023-08-05:
Oh. Interesting, I see `QK_BOOT` is now mapped to the `prgm` key in the default `keymap.c`.

lread, 2023-08-05:
And it actually works! Noice! 

@stapelberg This is very handy, especially for folks who put the screws back in their keyboards! I'm happy to add a note to the README, if you agree that makes sense.

stapelberg, 2023-08-06:
> @stapelberg This is very handy, especially for folks who put the screws back in their keyboards! I'm happy to add a note to the README, if you agree that makes sense.

Sure, I guess? This is a standard QMK feature, so a quick note with a link to their documentation should be sufficient.

lread, 2023-08-06:
Thanks for the reply! Because my motivation was to get my failing Advantage2 working again, I came to your project first first and then learned a bit about QMK. Maybe other folks come in knowing about QMK first. Anyway, your video really helped me with understanding how to solder things up and then flash. But, after watching it, maybe I missed it, but I naively never realized I could press a keyboard key to flash. I'll add a wee tip for other poor sods like me.