kinx-project/kint #11: Best way to connect thumb keys on k500

ecraven, 2020-08-24:
I've had a close look at the k500, I think I can probably unsolder the directly soldered connections for the thumb keys. You've kindly provided pin headers on your board, but is there some nice way to have a connector, not directly solder things together?
Thanks for any help with these questions ;)

stapelberg, 2020-08-24:
On the KB500, your best bet is to cut the cable (don’t bother unsoldering) and solder each pin to a pin header, like so:

[image]

ecraven, 2020-08-31:
great, that's what I've done, it works very well. Thanks!

closed 2020-08-31 by ecraven

jcbird, 2021-06-13:
Sorry to comment on a closed issue. But could you send a link to the exact part you used? Is this just a female header with the pins soldered to the pcb? Thank you so much!

stapelberg, 2021-06-13:
Yep, that’s all there is to it.

https://www.digikey.ch/product-detail/en/PPTC241LFBN-RC/S7022-ND/810162 is the part I used last time.

MrDeerly, 2023-03-07:
Sorry to dig out this old thread again, but there is still some unclarity for me. 

Are the female connectors mentioned above plugged into the [Würth 61301011121](https://octopart.com/61301011121-w%C3%BCrth+elektronik-18818159?r=sp) and soldered to the cable, or are the connectors soldered to the board and the cable just plugged in? 

In addition to that, would it be an option to solder the cable directly to the pin headers (i guess that's not what they're here for), or the board? 

Last but not least, do you have any tips on how to strip the cable for the thumb keys? :) 

Thanks in advance!

thatdutchguy, 2023-03-26:
In case someone else is wondering, I just went through this (kb500 usb) and I'll share my experiences:

> Sorry to dig out this old thread again, but there is still some unclarity for me.
>
> Are the female connectors mentioned above plugged into the [Würth [...]

They are soldered to the cable, then plug into the pin headers on the KinT PCB.
 
> > `[kinT] --soldered-- [male headers] --plugged-- [female headers] --soldered-- [stock cable] --soldered-- [PCB]`

Yes, I think that's the way it is intended to be used.

> > `[kinT] --soldered-- [female headers] --plugged-- [stock cable] --soldered-- [PCB]`

The wires are pretty thin, so I don't think this would work well.

> In addition to that, would it be an option to solder the cable directly to the pin headers (i guess that's not what they're here for), or the board?

> > `[kinT] --soldered-- [male headers] --soldered-- [stock cable] --soldered-- [PCB]`
> > OR
> > `[kinT] --soldered-- [stock cable] --soldered-- [PCB]`

I suppose this would work but it would be kind of cumbersome/harder than using the female headers.

> Last but not least, do you have any tips on how to strip the cable for the thumb keys? :)

I cut into the strip lengthwise next to each wire a bit, then used a simple wire stripper (see images below). It's not elegant but it works 🙂.

[image]
[image]

Hope this helps