kinx-project/kint #12: Debugging

ecraven, 2020-08-31:
Hello ;)
Just finished building one. I left the thumb pads disconnected, but connected everything else, cloned your firmware branch, flashed it. The keyboard shows up, however pressing any button does nothing at all. Is there some way to debug this?
Thanks for any help!

stapelberg, 2020-08-31:
You’re referencing a branch, so I assume you’re talking about the Teensy 3.6 variant.

Can you try flashing the firmware files referenced in https://github.com/kinx-project/kint#installing-the-firmware using https://aur.archlinux.org/packages/teensy-loader/?

Also, what is the dmesg output when you connect the keyboard?

stapelberg, 2020-08-31:
Did you solder in LEDs? If yes, try enabling Caps Lock using your currently working keyboard while the kint36 is connected, too: you should see both the LEDs on your current keyboard *and* the LEDs on the kint36 light up as well.

ecraven, 2020-08-31:
Just tried flashing the hex files, and those seem to work ;) great!

So the problem seems to be my compiling / flashing, not the hardware.

Caps-Lock light up the led on my current keyboard, but not the one on the new board. The LEDs have a bar on the underside, I put that bar in the same direction as the bar in the LED symbol on the silkscreen, is that correct?

ecraven, 2020-08-31:
dmesg -T output:
```[Mon Aug 31 22:25:48 2020] usb 2-3: new full-speed USB device number 31 using xhci_hcd
[Mon Aug 31 22:25:48 2020] usb 2-3: New USB device found, idVendor=1209, idProduct=345c, bcdDevice= 0.01
[Mon Aug 31 22:25:48 2020] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Aug 31 22:25:48 2020] usb 2-3: Product: "https://github.com/kinx-project/kint"
[Mon Aug 31 22:25:48 2020] usb 2-3: Manufacturer: "https://github.com/stapelberg"
[Mon Aug 31 22:25:48 2020] usb 2-3: SerialNumber: 0
[Mon Aug 31 22:25:48 2020] input: "https://github.com/stapelberg" "https://github.com/kinx-project/kint" as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:1209:345C.0024/input/input43
[Mon Aug 31 22:25:48 2020] hid-generic 0003:1209:345C.0024: input,hidraw4: USB HID v1.11 Keyboard ["https://github.com/stapelberg" "https://github.com/kinx-project/kint"] on usb-0000:00:14.0-3/input0
[Mon Aug 31 22:25:48 2020] input: "https://github.com/stapelberg" "https://github.com/kinx-project/kint" Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0025/input/input44
[Mon Aug 31 22:25:48 2020] input: "https://github.com/stapelberg" "https://github.com/kinx-project/kint" System Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0025/input/input45
[Mon Aug 31 22:25:48 2020] input: "https://github.com/stapelberg" "https://github.com/kinx-project/kint" Consumer Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0025/input/input46
[Mon Aug 31 22:25:48 2020] input: "https://github.com/stapelberg" "https://github.com/kinx-project/kint" Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0025/input/input47
[Mon Aug 31 22:25:48 2020] hid-generic 0003:1209:345C.0025: input,hidraw5: USB HID v1.11 Mouse ["https://github.com/stapelberg" "https://github.com/kinx-project/kint"] on usb-0000:00:14.0-3/input1
```

ecraven, 2020-08-31:
Flashing my own compiled version:
```> QMK_HOME=/home/.../qmk_firmware qmk flash -kb kinesis/kint36 -km default
Ψ Compiling keymap with make kinesis/kint36:default:flash

QMK Firmware 0.9.32
Making kinesis/kint36 with keymap default and target flash

arm-none-eabi-gcc (Arch Repository) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22792	      0	  22792	   5908	.build/kinesis_kint36_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/kinesis_kint36_default.elf                                                          [WARNINGS]
 | 
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:285: warning: memory region `flash4' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:293: warning: memory region `flash5' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:301: warning: memory region `flash6' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:310: warning: memory region `flash7' not declared
 | 
Creating binary load file for flashing: .build/kinesis_kint36_default.bin                           [OK]
Creating load file for flashing: .build/kinesis_kint36_default.hex                                  [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  22792	      0	  22792	   5908	.build/kinesis_kint36_default.hex

Copying kinesis_kint36_default.bin to qmk_firmware folder                                           [OK]
Teensy Loader, Command Line, Version 2.1
Read ".build/kinesis_kint36_default.hex": 22792 bytes, 2.2% usage
Found HalfKay Bootloader
Programming.......................
Booting
```
dmesg -T output (I've changed the names to Nex:Kint36, to see them):
```
[Mon Aug 31 22:26:27 2020] usb 2-3: new full-speed USB device number 32 using xhci_hcd
[Mon Aug 31 22:26:28 2020] usb 2-3: New USB device found, idVendor=16c0, idProduct=0478, bcdDevice= 1.03
[Mon Aug 31 22:26:28 2020] usb 2-3: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[Mon Aug 31 22:26:28 2020] usb 2-3: SerialNumber: 000B38F0
[Mon Aug 31 22:26:28 2020] hid-generic 0003:16C0:0478.0026: hidraw4: USB HID v1.11 Device [HID 16c0:0478] on usb-0000:00:14.0-3/input0
[Mon Aug 31 22:26:31 2020] usb 2-3: USB disconnect, device number 32
[Mon Aug 31 22:26:33 2020] usb 2-3: new full-speed USB device number 33 using xhci_hcd
[Mon Aug 31 22:26:33 2020] usb 2-3: New USB device found, idVendor=1209, idProduct=345c, bcdDevice= 0.01
[Mon Aug 31 22:26:33 2020] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Aug 31 22:26:33 2020] usb 2-3: Product: "Kint36"
[Mon Aug 31 22:26:33 2020] usb 2-3: Manufacturer: "Nex"
[Mon Aug 31 22:26:33 2020] usb 2-3: SerialNumber: 0
[Mon Aug 31 22:26:33 2020] input: "Nex" "Kint36" as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:1209:345C.0027/input/input48
[Mon Aug 31 22:26:33 2020] hid-generic 0003:1209:345C.0027: input,hidraw4: USB HID v1.11 Keyboard ["Nex" "Kint36"] on usb-0000:00:14.0-3/input0
[Mon Aug 31 22:26:33 2020] input: "Nex" "Kint36" Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0028/input/input49
[Mon Aug 31 22:26:33 2020] input: "Nex" "Kint36" System Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0028/input/input50
[Mon Aug 31 22:26:33 2020] input: "Nex" "Kint36" Consumer Control as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0028/input/input51
[Mon Aug 31 22:26:33 2020] input: "Nex" "Kint36" Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/0003:1209:345C.0028/input/input52
[Mon Aug 31 22:26:33 2020] hid-generic 0003:1209:345C.0028: input,hidraw5: USB HID v1.11 Mouse ["Nex" "Kint36"] on usb-0000:00:14.0-3/input1
```

stapelberg, 2020-08-31:
Can you try reverting the latest commit https://github.com/kinx-project/qmk_firmware/commit/1f84198750751dded9e12626b725df94aa33c875 in your local branch, and see if that fixes things?

stapelberg, 2020-08-31:
> Caps-Lock light up the led on my current keyboard, but not the one on the new board. The LEDs have a bar on the underside, I put that bar in the same direction as the bar in the LED symbol on the silkscreen, is that correct?

Yes, the LED marker marks the cathode, which is labeled as C on the kinT. It sounds like you did it correctly.

ecraven, 2020-08-31:
Reverting the last commit seems to work. I still have to try a custom layout, but I now get all kinds of keys ;)
Thanks for helping, if there's anything I can do to help debug this problem, please just ask!

ecraven, 2020-08-31:
Are those warnings about "region flash{4,5,6,7} not declared" relevant?

stapelberg, 2020-08-31:
> Reverting the last commit seems to work. I still have to try a custom layout, but I now get all kinds of keys ;)
> Thanks for helping, if there's anything I can do to help debug this problem, please just ask!

Great! Clearly, there is still some work to be done on proper EEPROM support. I have reverted the commit for now: https://github.com/kinx-project/qmk_firmware/commit/9206570c5cb681f18300a30933262f7658ec5ba2

I might get back to your offer when I next touch the EEPROM code :)

> Are those warnings about "region flash{4,5,6,7} not declared" relevant?

No, these regions aren’t required. Ideally we should fix those warnings eventually.

stapelberg, 2020-08-31:
I’ll close this for now as the board seems to be working now. If there are other problems, let’s track them in individual GitHub issues.

closed 2020-08-31 by stapelberg