kinx-project/kint #14: rules.mk: AUTO_SHIFT_ENABLE = yes does not compile.

ecraven, 2020-08-31:
If I enable auto shift keys in rules.mk, I get compilation errors:
```Linking: .build/kinesis_kint36_default.elf                                                          [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:285: warning: memory region `flash4' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:293: warning: memory region `flash5' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:301: warning: memory region `flash6' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:rules_memory.ld:310: warning: memory region `flash7' not declared
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-abort.o): in function `abort':
 | abort.c:(.text.abort+0xa): undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_kill_r':
 | signalr.c:(.text._kill_r+0x12): undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_getpid_r':
 | signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [tmk_core/rules.mk:306: .build/kinesis_kint36_default.elf] Error 1
```
It seems that auto_shift_keys link in something that might be in error?

stapelberg, 2020-09-01:
I cannot reproduce this:
```
% qmk clone -b kint36 kinx-project/qmk_firmware        
Cloning into '/tmp/autoshift/qmk_firmware'...
Submodule 'lib/chibios' (https://github.com/qmk/ChibiOS) registered for path 'lib/chibios'
Submodule 'lib/chibios-contrib' (https://github.com/ChibiOS/ChibiOS-Contrib) registered for path 'lib/chibios-contrib'
Submodule 'lib/googletest' (https://github.com/qmk/googletest) registered for path 'lib/googletest'
Submodule 'lib/lufa' (https://github.com/qmk/lufa) registered for path 'lib/lufa'
Submodule 'lib/printf' (https://github.com/qmk/printf) registered for path 'lib/printf'
Submodule 'lib/ugfx' (https://github.com/qmk/uGFX) registered for path 'lib/ugfx'
Submodule 'lib/vusb' (https://github.com/qmk/v-usb) registered for path 'lib/vusb'
Cloning into '/tmp/autoshift/qmk_firmware/lib/chibios'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/chibios-contrib'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/googletest'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/lufa'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/printf'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/ugfx'...
Cloning into '/tmp/autoshift/qmk_firmware/lib/vusb'...
Submodule path 'lib/chibios': checked out '313416b8fda90d9973a749a0a35970956852c286'
Submodule path 'lib/chibios-contrib': checked out 'cdb7fe0bf476fc044aee68d86f8b88faa10668d3'
Submodule path 'lib/googletest': checked out 'ec44c6c1675c25b9827aacd08c02433cccde7780'
Submodule path 'lib/lufa': checked out 'ce10f7642b0459e409839b23cc91498945119b4d'
Submodule path 'lib/printf': checked out 'd3b984684bb8a8bdc48cc7a1abecb93ce59bbe3e'
Submodule path 'lib/ugfx': checked out '40b48f470addad6a4fb1177de1a69a181158739b'
Submodule path 'lib/vusb': checked out '9a42d205eb60faca494ff4eabce8d59f0ec0fc7f'
Ψ Successfully cloned https://github.com/kinx-project/qmk_firmware to /tmp/autoshift/qmk_firmware!
qmk clone -b kint36 kinx-project/qmk_firmware  19,90s user 3,21s system 66% cpu 34,775 total

% cd qmk_firmware 

% echo AUTO_SHIFT_ENABLE = yes >> keyboards/kinesis/kint36/rules.mk

% qmk compile -kb kinesis/kint36 -km stapelberg
Ψ Compiling keymap with make kinesis/kint36:stapelberg
 

QMK Firmware 0.9.32
Making kinesis/kint36 with keymap stapelberg

arm-none-eabi-gcc (Arch Repository) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/kinesis/kinesis.c                                                              [OK]
Compiling: keyboards/kinesis/kint36/kint36.c                                                        [OK]
Compiling: keyboards/kinesis/keymaps/stapelberg/keymap.c                                            [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: tmk_core/common/chibios/eeprom_teensy.c                                                  [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_auto_shift.c                                             [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v7m.S                           [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.S                            [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
[... 106 lines cut ...]
Compiling: lib/chibios-contrib/os/hal/src/hal_eeprom.c                                              [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_timcap.c                                              [OK]
[... cut ...]
Creating binary load file for flashing: .build/kinesis_kint36_stapelberg.bin                        [OK]
Creating load file for flashing: .build/kinesis_kint36_stapelberg.hex                               [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  32792	      0	  32792	   8018	.build/kinesis_kint36_stapelberg.hex

Copying kinesis_kint36_stapelberg.bin to qmk_firmware folder                                        [OK]
(Firmware size check does not yet support cortex-m4 microprocessors; skipping.)

```

ecraven, 2020-09-01:
Can you try `-km default`? That fails for me :-/

stapelberg, 2020-09-02:
Thanks for the details.

I can reproduce this with a clean upstream checkout on another ChibiOS-based board:

```
% make matrix/noah:default                          
QMK Firmware 0.10.2
Making matrix/noah with keymap default

arm-none-eabi-gcc (Arch Repository) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: drivers/chibios/ws2812.c                                                                 [OK]
Compiling: keyboards/matrix/noah/matrix.c                                                           [OK]
Compiling: keyboards/matrix/noah/noah.c                                                             [OK]
Compiling: keyboards/matrix/noah/keymaps/default/keymap.c                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: tmk_core/common/chibios/eeprom_teensy.c                                                  [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight.c                                                                       [OK]
Compiling: quantum/rgb_matrix.c                                                                     [OK]
Compiling: quantum/rgb_matrix_drivers.c                                                             [OK]
Compiling: drivers/issi/is31fl3731.c                                                                [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: drivers/chibios/i2c_master.c                                                             [OK]
Archiving: .build/obj_matrix_noah_default/i2c_master.o                                              [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v7m.S                           [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.S                            [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/sendchar_null.c                                                          [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/chibios/suspend.c                                                        [OK]
Compiling: tmk_core/common/chibios/timer.c                                                          [OK]
Compiling: tmk_core/common/chibios/bootloader.c                                                     [OK]
Compiling: lib/printf/printf.c                                                                      [OK]
Compiling: tmk_core/common/bootmagic.c                                                              [OK]
Compiling: tmk_core/common/mousekey.c                                                               [OK]
Compiling: tmk_core/protocol/chibios/usb_main.c                                                     [OK]
Compiling: tmk_core/protocol/chibios/main.c                                                         [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: tmk_core/protocol/chibios/usb_driver.c                                                   [OK]
[... cut ...]
Make finished with errors
make: *** [Makefile:584: matrix/noah:default] Error 1
```

So I conclude that this issue is not kint-specific, and should be filed and fixed in https://github.com/qmk/qmk_firmware — could you please file an issue there?

https://github.com/qmk/qmk_firmware/issues/2089 is related, and perhaps the fix is to just add a few more syscall stubs?

For a workaround, you can comment out the `#include <stdio.h>` line and `snprintf()` call in `quantum/process_keycode/process_auto_shift.c`

stapelberg, 2021-06-30:
Not sure if this still happens, but either way, this should be reported at https://github.com/qmk/qmk_firmware as it has nothing to do with the `kint` controller.

closed 2021-06-30 by stapelberg