kinx-project/kint #17: trigger led's

jack0088, 2020-10-30:
Hey Michael,
first of all let me thank you for this project. It's amazing! I ordered all parts and the boards and have already build one. What a pleasurable and fun experience - it just works!

However I'd like to set my LEDs manually to create a Knight Rider effect (light scrolls left to right).
Ideally I'd like to have a function that I can all anywhere to turn on (or off) the light of a certain led.
I've found your initialization code inside kint36.c and played around with it, but I just can't turn on manually any of the led lights. Could you guide me a little into the right direction?

stapelberg, 2020-10-31:
> first of all let me thank you for this project. It's amazing! I ordered all parts and the boards and have already build one. What a pleasurable and fun experience - it just works!

Glad to hear everything works well for you!

> However I'd like to set my LEDs manually to create a Knight Rider effect (light scrolls left to right).
> Ideally I'd like to have a function that I can all anywhere to turn on (or off) the light of a [...]

My guess would be that the LEDs are overridden after you change them. Try commenting out the code which sets the LEDs based on the keyboard state: https://github.com/kinx-project/qmk_firmware/blob/kint36/keyboards/kinesis/kint36/kint36.c#L54-L91

But: maybe instead of implementing your own animation on just 4 LEDs, why not go one step further and control an LED strip?

QMK has support for that built-in: https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight, and there already is a knight rider mode animation: https://github.com/qmk/qmk_firmware/issues/1672

We have at least one person who got that feature to work on the kinT controller: https://github.com/kinx-project/kint/issues/10#issuecomment-679465162

afonsoguerra, 2021-06-11:
Can I ask for an update on the status of this?

At the moment I am still waiting for my boards to arrive, but I'm planning the code. I would instead like to use the LEDs as a layer indicator, so I could have up to 16 layers indicated using the 4 LEDs. The code referenced on kint36.c seems to have gone. So far I'm trying to make the keyboard look 'normal' so I ain't keen on the RGB mod, but looks easy enough.

stapelberg, 2021-06-11:
What update are you waiting for? AFAIK, the kinT controller LEDs work fine, and you can use the various QMK features that work with LEDs.

I updated the link in the comment above to point to https://github.com/qmk/qmk_firmware/blob/2e90ef05363893f3aa49c99af408e774609aa97a/keyboards/kinesis/kint36/config.h#L62-L66, where QMK configures which pins to use for the status LEDs.

afonsoguerra, 2021-06-11:
ah, awesome. I was just thrown off by the link above. no updates needed, seems this will work out of the box.

stapelberg, 2021-06-30:
Looks like this is resolved, closing the issue.

closed 2021-06-30 by stapelberg