kinx-project/kint #18: [Teensy3.6] Some keys are not recognized on MacOS

Sanic, 2020-12-25:
See also https://github.com/kinx-project/kint/issues/16#issuecomment-748665978

It seems that the Teensy 3.6 version of this controller might behave weirdly on MacOS:
  - Some keys are not detected properly (For example the thumb keys for delete, enter, etc.)
  - Keys are only activated if another key close to it is pressed (same column?)

Is there somebody who uses the teensy 3.6 variant of this controller with a mac?

stapelberg, 2020-12-29:
I just tested my Teensy 3.6 Kinesis on a MacBook Air M1 running macOS Big Sur 11.0.1, and all keys seem to work perfectly fine.

You mentioned that you did a separate build with a teensy++ that worked. I assume you have been using both controllers in the same keyboard, so we can likely exclude keyboard issues.

Can you double-check all soldering joints on the affected board? Applying some extra flux in liquid form, if you have it, or just a little fresh solder and re-heating each solder joint will usually help.

stapelberg, 2021-06-30:
I’ll close this as there were no other reports of this issue, and I couldn’t reproduce it.

closed 2021-06-30 by stapelberg