kinx-project/kint #19: USB hub possible?

impaktor, 2020-12-31:
## Intro
Thanks for your amazing work on this project!

My background: I'm using a kinesis Advantage 2 (KB600-SE), and I have previously built a [Dactyl-ManuForm](https://github.com/tshort/dactyl-keyboard) with arduino + QMK, but beyond that I don't have any electrical engineering experience and thus zero kicad/pcb work, thus my following questions below.

## 1. USB hub?
The only function I miss in my Kinesis 2, is a USB-hub (ports both for internal and external use), so I'd be very interested in how/if there is any possibility to add that to the kinT?

I know you [experimented](https://michael.stapelberg.ch/posts/2018-04-17-kinx-usb-hub/) with this a few years ago, for the older board (kinx) you were working on then. I think you came to the conclusion that the sum of connected USB-devices must be below 500 mA. Would that still be a limiting factor if it were to be re-implemented on this board? Is there space on the board for it?

## 2. Speaker?

The factory board comes with a piezo buzzer, to give audio feedback, possibly to make up for the switches not being clicky, I guess. I find I actually use it now that we're all working from home (thus not disturbing my colleagues), I also used it when learning to touch type when I was new to the kinesis. Are there any free pins on the teensy for adding this, and could the traces be added for this on the board?

## 3. Github discussion?

I saw github recently launched a [discussion page](https://docs.github.com/en/free-pro-team@latest/discussions), which can be enabled, if you prefer to keep issue tracker devoted to only bugs/issues, and discussions/forum type posts elsewhere

## Conclusion

I'm half itching to make this board (for QMK and green LEDs > blue LEDs!), but I want to hear your thoughts on my questions above, if you have time to answer them.

Cheers!

stapelberg, 2020-12-31:
> Would that still be a limiting factor if it were to be re-implemented on this board?

Yes: the limit is how much power computers are willing to deliver via USB, so unless you add a separate, external power supply (like an active USB hub does), 500mA is all we got.

> Is there space on the board for it?

Yes in theory, but no in practice: keeping the USB hub on a separate board is the better choice. That way, you can reason more easily about interference, which becomes an issue with USB 2.0 (minimum one would expect in a hub these days), and even more so with USB 3.

My recommendation is to just build the existing kinX USB hub board, which is known to work. Designing your own will be very time-intensive. The KB600 requires modifications to the case to expose the ports, so that’ll be enough work :)

> Are there any free pins on the teensy for adding this, and could the traces be added for this on the board?

This depends on the teensy model you’re using: all teensys that have enough pins for the LEDs (Teensy++, Teensy 3.5, Teensy 3.6, Teensy 4.1) have enough pins for additional peripherals. Check out the schematic for more details: https://github.com/kinx-project/kint/blob/master/schematic-v2020-06-30.pdf

As for adding the trace/peripheral itself, I think that would best be done in a separate modification for now. It’s not clear to me which peripherals are the most popular/useful, and every change to the board requires careful testing etc., so I’d like to keep the board as minimal as it is for now. The design of the project should make it easy for you to add your desired peripherals, be it speakers or LED strips, or… :)

> I saw github recently launched a [discussion page](https://docs.github.com/en/free-pro-team@latest/discussions), which can be enabled, if you prefer to keep issue tracker devoted to only bugs/issues, and discussions/forum type posts elsewhere

I haven’t used GitHub discussions yet. Have you used them and found the experience good? We can experiment with turning them on, I suppose.

impaktor, 2021-01-01:
Thanks for swift answer, that straightened out some question marks I had.

> I haven’t used GitHub discussions yet. Have you used them and found the experience good?

I have not used them either, as in other projects I'm involved in we already have forums going hosted elsewhere. Perhaps my suggestion was a justification to myself for opening an issue that is not a bug report, something I'm in general against doing.

closed 2021-01-07 by impaktor