kinx-project/kint #22: Teensy++ 2.0 positioning on board question

dky, 2021-01-09:
Hello @stapelberg,

Thank you publishing this, this actually is such an improvement because the board is actually positioned properly. I've been using a variant of your board and it needs to get soldered in backwards which is so weird (Also wondering if this is why by boards have been dying so frequently. I've lost about 3 of them - they just randomly stop working). 

Question: I have an older teensy++ 2.0 and wanted to clarify with you the positioning before I started soldering:

[image]

I wanted to know if this was where the teensy++ should sit. I'll also follow your directions below once done.

```
Soldering instructions for the Teensy++ 2.0
Follow the instructions for the Teensy 3.x or 4.x above, but:

Do not connect pin 7, pin 15 and pin 16. These are marked with an x on the kinT.

An easy way to do this is to remove the corresponding pins from your pin header with pliers.
Close the solder jumpers JP1, JP2, JP3. These will remap pin 7, pin 15 and pin 16 onto pins that can be used with the Teensy++ 2.0.
```

Afterwards once this is confirmed.

Thank you!

dky, 2021-01-09:
Another question is when you say "Close the solder jumpers JP1, JP2, JP3" does that mean create a bridge between the following on the back of the unit?

[image]

I'm planning to put a two wires to bridge JP1 + JP2 + JP3 will that work?

dky, 2021-01-09:
Here's a visual diagram of what I plan to do for the bridge:

[image]

Please let me know if this will work!

stapelberg, 2021-01-09:
> (Also wondering if this is why my boards have been dying so frequently. I've lost about 3 of them - they just randomly stop working).

One theory might be ESD (electro-static discharge). A friend of mine told me that directly accessible switches (like key switches) should ideally be ESD-protected, but I haven’t seen this much on other keyboard projects.

> I wanted to know if this was where the teensy++ should sit.

Yep, that looks correctly positioned to me.

> Another question is when you say "Close the solder jumpers JP1, JP2, JP3" does that mean create a bridge between the following on the back of the unit?

No: these are 3 separate solder jumpers. To close each one, just apply solder within each of the white squares, creating a connection between the two golden pads within the square.

Here’s an example (apologies for the dusty PCB):

[image]

dky, 2021-01-09:
@stapelberg Thanks for the quick response! I got it. I'll poke at this shortly. Thank you for clarifying what that bridge meant I was a bit confused since I'm still pretty much a novice with hardware.

dky, 2021-01-09:
I'll also keep in mind the ESD thing. I was thinking it was the quality of the Teensy controllers but will be mindful of ESD as well.

closed 2021-01-09 by dky