kinx-project/kint #25: LED issue with Teensy 4.1

gottwald, 2021-03-13:
I've built the board with a Teensy 4.1 and used the firmware form your qmk_firmware fork (https://github.com/kinx-project/qmk_firmware) and kint41 branch (e2f82ab2dac).
I've used the exact components from the guide.

Mostly stuff seems to work but none of the LEDs do.

I've done some measurements and I'm a bit confused by the results:
- I've got 3V on the `3V3_2` pin of the cross row. I also didn't forget to solder it.
- I've got 3V on all anodes
- I checked that I have no shorts between the LED Pins `12`, `24`, `25`, `26` 
- All measurements were done with several ground pins (`GND`, `GND mounting pad`, `GND 3`)
- Can't make the LEDs light up with my multimeter (weirdly shows `.0L` in forward and reverse bias. I use a Brymen BM257s which IIRC has a 1V range for diode testing which might be the reason). But I can make them all light up with some help from a 3V battery
- Checked the resistors, all measure 10k.

Here's where it becomes strange:
- As said above I can always measure 3V on the anode
- But I also can always measure 0.9V on the cathode. Regardless of CAPS_LOCK being on or off for example.
- I can also measure the 0.9V before and after the resistor.
- I also got the 0.9V on the LED Pins `12`, `24`, `25`, `26`.
- During those measurements the LED lights up a tiny bit (barely visible)

The rest of the keyboard seems to work fine as I'm typing this issue on it.
The only thing that doesn't work is waking up my mac from sleep with this keyboard, but I'm guessing this is more of a firmware thing.

Any idea what could be wrong?

stapelberg, 2021-03-15:
Give https://github.com/kinx-project/qmk_firmware/commit/c61d90b80415343d86b9c0c1b659823aaaee47c9 a try — I think the LEDs just weren’t wired up before.

But, the GPIO functionality is working now, so I think the wiring should be okay. In a quick test, this made the Caps Lock LED work for me on my Teensy 4.1 board.

gottwald, 2021-03-15:
Thank you! Works like a charm.
I was only able to verify CAPS_LOCK due to my layout but re-measured the pins and now the measurements on all LED pins make sense so I'm pretty sure the others are also OK.

closed 2021-03-15 by gottwald