kinx-project/kint #27: enhancement: make design classify as simple at aisler?

stapelberg, 2021-04-01:
Aisler recently introduced a new, cheaper tier: https://aisler.net/blog/introducing-budget-manufacturing-the-most-economic-choice-on-the-planet

Our current artifacts (v2020-06-30) show up on the aisler web interface as “complex”, which makes it not qualify for this new, cheap tier:

Dimension / Size | 107.95mm x 86mm
-- | --
Design Rule Rating | Complex
Surface finish | ENIG
Layer Count | 2
Minimal drill diameter | 0.4mm
Elongated holes count | 0

See https://aisler.net/help/design-rules-and-specifications/design-rules for aisler’s design rules.

AFAICT, we fulfill 2 of 4 criterions: our minimal drill diameter is 0.4mm, and we have 2 layers.

Is the problem plated slots? The milling tool size? Something else entirely?

If we can, it would be nice to make our design qualify as a simple design.

stapelberg, 2021-04-13:
The problem seem to be the cut-outs:
[image]

As soon as I remove them, Aisler classifies the design as simple instead of complex.

Now the question is whether there is an alternative way to design the cut-out that qualifies as simple.

cincodenada, 2021-04-25:
So I started to look into this, wondering if connecting the holes to the board outline would help, but then upon looking at the guidelines, I think this is what we're getting nicked for:

> smallest necessary milling tool for cutouts | 2.4 mm

The cutouts are, frustratingly, 2.1mm and change wide in the short dimension. I widened those slots just a tad so they were over 2.4mm, and AISLER did indeed judge it to be Simple:

[image]

I thought I'd have to make them drill slots, but apparently they're happy with the squares as long as they're at least 2.4mm wide. Obviously this makes it more able to jostle around, but that seems acceptable, although it's a bummer to have weird cutouts just for Aisler.

To that end, I made even weirder cutouts that pass as Simple, that are little zigzags so that the width is always 2.4mm, but half of each side of the slot will still be even with the posts. That's in the latest revision on my Aisler test project: https://aisler.net/p/ZLFAZHWW

[image]

I have no clue what those slots will come out like in actual production, but they at least pass whatever check they have, and I imagine if anything they'll end up being wider than designed, which is probably fine?

stapelberg, 2021-04-25:
Thanks for figuring this out! 

> I have no clue what those slots will come out like in actual production, but they at least pass whatever check they have, and I imagine if anything they'll end up being wider than designed, which is probably fine?

Yes, that’s fine.

> To that end, I made even weirder cutouts that pass as Simple, that are little zigzags so that the width is always 2.4mm, but half of each side of the slot will still be even with the posts. That's in the latest revision on my Aisler test project: https://aisler.net/p/ZLFAZHWW

Cool! I like that idea :)

As mentioned in your pull request #30, if you could split out this fix into a separate pull request, that’d be great. I will order a test run at Aisler today if you can send the changes :)

stapelberg, 2021-04-25:
> but they at least pass whatever check they have

By the way, my current theory is that they cut the board outline with a 2.4mm drill bit, so unless the slots are at least 2.4mm high, they need to switch to a different drill (or perhaps even different machine).

closed 2021-04-25 by stapelberg

cincodenada, 2021-04-25:
Yeah, that's my assumption as well, which is why I'm curious about what it will look like - if they're cutting with a 2.4mm drill bit, the corners of the zigzag might not be very rectangular!

stapelberg, 2021-04-25:
> if they're cutting with a 2.4mm drill bit, the corners of the zigzag might not be very rectangular!

Looking at their previews, I’m convinced they cut with a drill, and the corners will not come out rectangular. In fact, looking at the boards I ordered from OSH Park, they also clearly use a drill and non-rectangular holes are the result. Doesn’t seem to be a problem for fitting the board in the Kinesis :)

I have tagged revision v2021-04-25 and uploaded it to aisler: https://aisler.net/p/NLNHRLZN

I also ordered 3 PCBs, let’s see how they come out :)

stapelberg, 2021-04-29:
> I have tagged revision v2021-04-25 and uploaded it to aisler: https://aisler.net/p/NLNHRLZN
> 
> I also ordered 3 PCBs, let’s see how they come out :)

Just received the boards this morning, and I am happy to report that the cut-out fits perfectly fine! :)

[image]

commit 48328a4c (2021-04-25): Widen slots for compatibility with Aisler (#31)

Aisler is a board service that introduced a budget service for "Simple"
boards. KinT almost met the requirements, but these slots were too
narrow. These zigzags make it so that the slots are always 2.4mm wide,
but should still contact the posts and stabilize the board.

fixes #27