kinx-project/kint #28: status prints dvorak even when qwerty and initial keymap doesn't persist after unplug

cartazio, 2021-04-19:
i'm using 

```
commit 0a9e18fae13654ab6d8d85a9867fc10a4261103f (HEAD, origin-primary/develop)
Merge: 924b9fcf0 eb7e668eb
Author: QMK Bot <hello@qmk.fm>
Date:   Sun Apr 18 22:40:16 2021 +0000
```

and heres what status prints:
```Firmware> QMK 0.12.35-312-g0a9e18, 2021-04-18-20:15:11
Keyboard> kinesis/kint36
Keymap> kzar
Layout> DVORAK
Thumb keys mode> MAC
NKRO> Enabled
Debug> Disabled
```

some questions:
1) why cant i persist the mapping /layer state?  2) failing that, how do i patch the code from that commit to start with the mac layer being the intitial one?

thankx :)

stapelberg, 2021-04-19:
The settings not being persisted likely is a duplicate of https://github.com/kinx-project/kint/issues/8 

I’m not sure about the wrong status, though, that might be a QMK issue.

Perhaps someone else reading this issue has any ideas?

cartazio, 2021-04-19:
So a simple fix for me would be just to change the layers order somehow so mac comes first. How would I change the code at that point for that?

stapelberg, 2021-04-19:
I don’t use QMK layers, I use the [NEO layout](https://neo-layout.org/) which comes with its own layers.

Your question might be better asked in the QMK project itself, not in this specific controller project :)

stapelberg, 2021-05-14:
https://beta.docs.qmk.fm/using-qmk/software-features/feature_layers seems to be the documentation from QMK regarding the layers feature.

The layers you’re using are specific to the kzar QMK keymap, so perhaps @kzar can shed some light on this?

kzar, 2021-05-14:
> The settings not being persisted likely is a duplicate of #8

Yea, I think that's right. I could never get the layers to persist at all. If you ever get it working with the Teensy 3.6 @stapelberg let me know and I'll test that the various layers in my keymap persist correctly. It's possible I'll need to make changes too.

> So a simple fix for me would be just to change the layers order somehow so mac comes first. How would I change the code at that point for that?

You would edit the file `keyboards/kinesis/kint36/keymaps/kzar/keymap.c`. The different thumb modes are set up as different layers, with the exception of the default WIN mode which is built into the default QWERTY layer. The simplest way to bodge it, would probably be to see where [the thumb keys for the QWERTY layer are set](https://github.com/qmk/qmk_firmware/blob/develop/keyboards/kinesis/kint36/keymaps/kzar/keymap.c#L197-L200), and adjust them [to match the MAC layer](https://github.com/qmk/qmk_firmware/blob/develop/keyboards/kinesis/kint36/keymaps/kzar/keymap.c#L249-L252).

> status prints dvorak even when qwerty

Huh, I just tested it and you're right. That's weird since I obviously had this working correctly previously. I will investigate and attempt to fix that when I get a chance. Thanks for letting me know!

stapelberg, 2021-06-30:
Looks like the wrong status was fixed with https://github.com/qmk/qmk_firmware/pull/12895

The persistence issue will be fixed once QMK merges the EEPROM support PR: https://github.com/qmk/qmk_firmware/pull/12947

I’ll close this issue, please follow the PR if you’re interested.

closed 2021-06-30 by stapelberg