kinx-project/kint #29: Documentation clarification re: Teensy++ 2.0 @ 5V

cincodenada, 2021-04-21:
Hello! Thanks for putting this together, I'm building my first kit now. I decided it would be a good resting place for my old Teensy++ 2.0, and thus have a question about docs - is the Kinesis 5V tolerant (or perhaps 5V originally?) I just want to make sure I don't break anything by plugging in an unmodified 5V Teensy++ 2.0, because the only mention of voltage in the README was the LED voltage at 3.3V.

Thanks!

stapelberg, 2021-04-21:
Yep, the key switches and diodes in the Kinesis (the only electronics components aside from the controller board) are 5V tolerant. 

I think the older KB500 controller might even have been running at 5V, but I can’t find a high-enough resolution picture to confirm which microcontroller it used…

cincodenada, 2021-04-21:
Rad, thanks! And actually, now that I think of it I'll be cracking my KB500 open soon, so I'll make sure to take some good photos and confirm :)

stapelberg, 2021-04-21:
Great! Good luck with the modification, and I’m always happy to see a picture of a successful modification :)