kinx-project/kint #32: Use Neopixel LEDs (1 pin) like the SK6812mini-E?

stapelberg, 2021-04-29:
As suggested in https://twitter.com/deBaer/status/1387823539273801729

Looks like https://octopart.com/2686-adafruit+industries-68087195?r=sp is the corresponding part.

Changing the layout would make the kint36 and kint41 builds 4.5 USD more expensive, but add LED capability to the kint40.

I wonder if a hybrid layout is possible…

stapelberg, 2021-04-29:
Clarified on twitter: it’s a different SKU: https://twitter.com/deBaer/status/1387827578753490952

But that one unfortunately is not available on Digikey/Mouser, so we’ll need to wait with this change until they are.

tamsky, 2021-07-05:
Showing as "in stock" at Digikey [1], but not on "manufacturer" site [2].

[1]: https://www.digikey.com/en/products/detail/adafruit-industries-llc/4960/14302512  (reverse mount style)
[2]: https://www.adafruit.com/product/4960

stapelberg, 2021-07-05:
With the current supply chain situation, I don’t think we can switch to neopixel LEDs.

We should keep an eye on the distributor / market situation once things are more normal (in a few years, I suppose).

(If people want to experiment until then, feel free, just trying to set expectations for the time frame until I’ll feel confident to accept such a change into the project.)