kinx-project/kint #34: Fix for soldering every pin of teensy++ ?

mgsloan, 2021-05-16:
I purchased a v2020-06-30 kinT on ebay, and unfortunately it does not have the annotations specifying to see soldering instructions, or that the jumpers are for teensy++ compat.  I have already soldered every pin of a teensy 2.0++.

I'm not confident in my ability to desolder the teensy in order to remove the pins and resolder.  What will go wrong if I try to use it with these pins also soldered?  Simply not working at all?  Can I just cut the traces to these pins instead?

Thanks!

mgsloan, 2021-05-16:
Oof, I see that cutting traces would only work for pin 16..

stapelberg, 2021-05-16:
> I purchased a v2020-06-30 kinT on ebay, and unfortunately it does not have the annotations specifying to see soldering instructions, or that the jumpers are for teensy++ compat.

Why is it lacking these annotations? Might be worth filing a complaint with the seller.

> Oof, I see that cutting traces would only work for pin 16..

Why would it not work for the other pins? Cutting traces and adding jumper wires should work for all the pins, I think.

mgsloan, 2021-05-16:
Ahh, I had not thought of adding jumper wires (my earlier reference to
jumpers was about the jumper pads).  Thanks!!!  Will try that soon.

I'm not sure why the annotations aren't there, I assume it's due to the PCB
layout being nearly a year older than the repo version.  I've sent the
seller a message about it.  For reference, it's this ebay posting -
https://www.ebay.com/itm/274653421480 - very convenient to be able to order
a board with almost all of the components soldered!

mgsloan, 2021-05-16:
Ah, I've heard back from the seller and the PCB vendor just didn't do the
silk screening on that side of the board.  The seller very helpfully
offered to fix the issue for me, but I think I will probably just implement
the hack myself, since I have some background in electronics tinkering.

stapelberg, 2021-05-16:
Alright! Good luck with the fix :)

closed 2021-05-16 by stapelberg

mgsloan, 2022-03-08:
I finally got around to trying to get this working, since I now have need for multiple keyboards (now have an office setup again).  My desoldering skills just are not up to removing the teensy and starting again (I just tried heat gun and iron though).  So, I tried the trace cutting + jumpers approach and it worked!

Perhaps this gorey image of my hackery will help someone.

[image]

Note that sometimes my trace cutting scraping went a bit off track and over other traces (but just surface scratch, so they still work), so don't follow this as a direct guide for what traces to cut.  The jumpers should be clear, though.  Naturally, a multimeter is handy to ensure the trace is actually cut.  Can also check if the cutting process accidentally grounded the pins - looks like I accidentally grounded pin 16, but it doesn't seem to matter.