kinx-project/kint #36: Teensy 3.6 (kint36): implement audio support in QMK/ChibiOS

stapelberg, 2021-05-20:
QMK’s quantum/audio/driver_chibios_dac_basic.c is tied to the STM32 DAC implementation.

To make audio work on the Teensy 3.6, we’d need to update that file to work with whichever ADC/DAC hardware is available on the MK66F18, which likely also requires a ChibiOS-Contrib contribution.

I have no use for audio so I don’t plan to work on this myself. Contributions welcome!

Timost, 2021-10-05:
Hi @stapelberg .

First of all, I would like to say  a huge thank you for your work :pray: :pray: :pray:.

Thanks to you, I've been able to hack my current keyboard using a Teensy 4.0. [repo not up to date](https://gitlab.com/Timost/thinkpad-kbd-qmk)

Lately I've been trying to get QMK audio working on the Teensy 4.0 without success.
I found this issue and my guess is that it also applies to teensy 4.x, could you confirm ?

I'm not really familiar with Chibios so, sorry for Stapelbergasking you directly instead.

stapelberg, 2021-10-06:
Yeah, I think QMK’s implementation only works with the STM32, and all other micro controllers are unsupported.