kinx-project/kint #37: Support for the (upcoming) Kinesis Advantage360

stapelberg, 2021-05-26:
See e.g. https://twitter.com/kinesisergo/status/1397335511694381058?s=21

That keyboard is not yet released, but I’m filing this issue anyway so that people can contribute details or subscribe for status updates

aaronjensen, 2022-08-24:
@stapelberg are you still thinking about working on this? I'd be interested in a fully-wired 360 w/ QMK personally, but no idea how feasible that is or if you're going to be looking at ZMK now 😄

stapelberg, 2022-08-24:
I received my 360 (non-pro) recently, but haven’t had a chance to open it up yet. I’m not quite decided as to what I’ll do, or even what is possible :)

aaronjensen, 2022-08-24:
Ah, good stuff. I'd be curious to see a teardown from you some day :)

aaronjensen, 2022-08-29:
@stapelberg is it possible and have you thought at all about flashing the Pro version with QMK and figuring out a way to connect them?

stapelberg, 2022-08-29:
Kinesis says the linking between the modules is always wireless on the Pro: https://twitter.com/zekjur/status/1470864813387616258

I can’t say yet what’s possible, as I don’t have a Pro version yet — they haven’t shipped yet.

stapelberg, 2022-08-30:
Here’s a video of my first look at the Kinesis Advantage 360: https://www.youtube.com/watch?v=jHXmW87_WpY

brandondees, 2022-09-15:
got mine in tonight. let me know if there are specific questions i can help answer. 

not sure what the benefit of converting this to QMK would be, but i'm not too familiar yet with ZMK or how it differs from QMK.

jfrankl, 2022-10-04:
@stapelberg what is your evaluation of the 360 Pro in terms of latency? I am considering buying a 360 Pro because QMK/ZMK firmware is needed for my layout, but wireless isn't important, and my one concern is lag.

stapelberg, 2022-10-04:
I haven’t done an end-to-end latency measurement yet, but what I can say is that it ships with 10ms USB polling interval, which you can reduce to 1ms: https://twitter.com/zekjur/status/1572327757849174023

stapelberg, 2022-10-04:
Oh, thanks for letting me know and for getting this merged! :)

aaronjensen, 2022-10-04:
Ah, I didn't get it merged, it was already upstream. And no problem.

AI-Guru, 2023-06-11:
Interesting! Just dropping by to say that I would love to see the possibility of QMK on Kinesis 360 Advantage (non-pro).

stapelberg, 2023-06-12:
Someone would need to do the reverse-engineering for it. 

With the 360 hardware, you can’t just swap in a new controller — the PCB has a much more complicated form factor, and a lot more intricate components on it. The only possible solution I see is to keep the PCB as-is and flash a new firmware. But I don’t know which micro controller they used and how open it is.

We’re talking about a lot of development work, with the additional “problem” that most people who care about free firmware just buy the 360 pro instead… Personally, I have no plans to do anything about the 360 non-pro.

pepsiplease69, 2023-10-15:
I personally am looking for the possibility of using qmk on the non-pro version because of the auto-shift feature in qmk. 

The stock firmware has some tap&hold functionality but it’s not recommended for the alphanumeric keys.

I’ve opened a ticket with kinesis support to ask for the auto-shift feature in the form of a firmware update. Let’s see what they say. 

I’ll send updates as anything develops. 

I’ve easily been able to desolder the switches and install my favorite boba u4 so, at least that was good about the design.

Malikazz, 2025-11-11:
Hello I have the 360 non pro and enjoy the form factor quite a bit. The firmware on the other hand I find to be quite crap. The keyboard crashes and restarts on the regular for me from a known bug that seems to have no chance of being solved by the vendor. I am not a big hardware hacker but I am a competent programmer, would anyone here be able to point me at a book or other resources for getting started on customizing the 360. I have several micro controllers laying around that maybe I could use for testing.