kinx-project/kint #39: Teensy 4.1 (kint41) - Two key inputs with single key press

tsharky, 2021-06-25:
Hi,

first of all thank you very much for your great work and support on the kint project.

I have two kinesis advantages 2 and have built three kint pcbs in total. Two of them with Teensy 4.1.
My advantage with the cherry mx quiet red switches works totally fine and I could develop a bone like layout (close to neo) for it.

But when I use the same firmware build on my second advantage with cherry browns, I get a weird behavior when typing mostly on my pinkies and ring fingers.

1 out of 10 I see two key inputs with a single key press. Seems to be related to the 'bump' of the tactile switches.

Is there a way to increase the debounce time? Or is there another way to fix this?
This behavior is really disturbing, but I cannot use the native kinesis controller with the bone layout.

greetings
Tim

stapelberg, 2021-06-25:
Which version of the firmware are you using?

https://github.com/kinx-project/qmk_firmware/commits/kint41-upstream-ref already contains a debounce time of 20ms, which should work well. Likely you’re on an older version?

tsharky, 2021-06-25:
I forked your kint41 branch ~2 weeks ago.
But it is set to 5 ms. I will change it to 20 and test it.
I will comment or close this issue!

closed 2021-06-25 by tsharky

stapelberg, 2021-06-25:
I take it bumping the debounce time to 20ms fixed the issue for you, yes?

tsharky, 2021-06-25:
yes, 20 ms debounce time fixed it.