kinx-project/kint #40: Teensy 3.6 - Keyboard freeze when interacting with LEDs?!

tsharky, 2021-06-25:
I use the default keymap and kinesis/kint36 keyboard of the upstream qmk firmware.

The keyboard works fine until I press caps lock, scroll lock or num lock.
The LEDs light up, but there are no more key inputs possible. The keyboard is unresponsive.

I need to deactivate all locks with a 2nd keyboard and need unplug + reconnect the keyboard, to make it work again.

If I replug with an active lock state, it will not work.

stapelberg, 2021-06-25:
Can you add which branch and git revision of the upstream qmk firmware you’re using please?

tsharky, 2021-06-25:
https://github.com/qmk/qmk_firmware/tree/224dcac08fe65cd4b4ab275cac4aea1379c3528c

Is that what you need? Its sha 224dcac08 'from' the master branch.

stapelberg, 2021-06-25:
Yep. Can you try if this issue is reproducible with the `develop` branch, too? Please let me know the precise revision of the `develop` branch you’re testing with, too.

tsharky, 2021-06-25:
How do I build from develop? oO

I checked out develop (https://github.com/qmk/qmk_firmware/tree/0ef0c03a4649a3b66c68587f1e25333d06887a82) and did 

`git submodule update --init --recursive`

to update git submodules. Is that correct? Never worked with submodules before.

Branch is up to date with remote and working tree is clean.

If I try to compile
`qmk compile -kb kinesis/kint36 -km default`

compile is not an argument anymore..

> usage: qmk [-h] [-V] [-v] [--datetime-fmt DATETIME_FMT] [--log-fmt LOG_FMT] [--log-file-fmt LOG_FILE_FMT]
>            [--log-file-level {debug,info,warning,error,critical}] [--log-file LOG_FILE] [--color]
>            [--no-color] [--unicode] [--no-unicode] [--interactive] [--config-file CONFIG_FILE]
>            {config,clone,env,setup} ...
> qmk: error: argument {config,clone,env,setup}: invalid choice: 'compile' (choose from 'config', 'clone', 'env', 'setup')

Did a clean clone
`qmk clone -b develop`

same result.

If I checkout master again, it works as usual.

stapelberg, 2021-06-25:
> to update git submodules. Is that correct? Never worked with submodules before.

I use `make git-submodule`

> If I try to compile
> `qmk compile -kb kinesis/kint36 -km default`

Try `make kinesis/kint36:default`

tsharky, 2021-06-25:
I don't get it...

`make kinesis/kint36:default`

> QMK Firmware 0.13.14
> ERROR: Can not run "qmk" command!
> 
> Please run util/qmk_install.sh to install all the dependencies QMK requires.
> 
> make: *** [Makefile:512: kinesis/kint36:default] Error 1

`which qmk`

> /home/tschaake/.local/bin/qmk

Mentioned install script change nothing.

Switching back to master works.

stapelberg, 2021-06-25:
Yeah, I had to manually delete qmk from my system and re-install it, too.

tsharky, 2021-06-25:
Qmk reinstall helped. Thanks.

I did some further testing and it seems to be an issue on my side.
It happens when I plug in the qmk advantage as a 2nd keyboard (no difference between kint36 and kint41).

Tested it on my laptop too. There is no keyboard freeze.

closed 2021-06-25 by tsharky

stapelberg, 2021-06-25:
> It happens when I plug in the qmk advantage as a 2nd keyboard (no difference between kint36 and kint41).

Does adjusting https://github.com/qmk/qmk_firmware/blob/master/keyboards/kinesis/kint36/config.h#L60 to 2 or 3 help by chance?