kinx-project/kint #41: Teensy 4.1: Faulty behavior with recent windows 10 update

tsharky, 2021-08-03:
Hey!

Both my kint41 and my kint36 do not work properly with the recent windows 10 update (weather thingy in the task bar).
I noticed following issues:
- keyboard detection failed from time to time
- input is delayed/laggy
- 'fast' typing results in spamming the same key endlessly. Keyboard is freezed and needs to be re-plugged.

I built them from upstream https://github.com/qmk/qmk_firmware/tree/34de7ca224d613e1ae19a45860e27c15d40254dd.

Current state is unusable in windows. I checked it on two machines. The kints worked fine until the recent windows update.
I have no issues on my linux machine.

Would be nice if you could take a look or check it on a windows machine.
Which information can I  provide to help solving this issue?

Greetings!

stapelberg, 2021-08-03:
Can you report this issue at the QMK repository please? It does not sound hardware-specific, but rather like a firmware issue.

tsharky, 2021-12-22:
I guess the problem was a cheap USB cable.
Don't understand why it worked properly on Linux but not on Windows.

closed 2021-12-22 by tsharky