kinx-project/kint #42: older Kinesis classic (KB333) compatibility?

stapelberg, 2021-10-07:
I have gotten two questions regarding compatibility with the older kinesis advantage classic (KB333), which predates the KB500 and KB600 models.

The brief answer is that the kinT is currently not compatible with the older classic version.

The long answer is:

Electrically it is compatible, mechanically it is not compatible. The biggest difference are the connectors to the left and right key wells. Newer kinesis revisions use FFC connectors (FFC = flat form factor cables), but in the older revisions, the key well PCBs are directly connected into the controller (no cable).

Unfortunately I don’t know the part number for the required connector. In the humblehacker blog article, the author says they de-soldered the original connector from the original controller. (So if the keyboard is broken and you’re feeling adventurous, that might be an option.)

If you can figure out which part number one needs, you only need to replace it in the kinT controller design. The footprint should be identical or similar.

The only bit that’s worrying me is the physical position on the controller. With the FFC connectors you have a little wiggle room, but I think when connecting the PCBs directly, everything needs to fit much more precisely. Possibly you need to move the connector to the left or right in the kinT controller design, but that might require moving traces as well…

I hope this gives an impression of the kind and amount of work required to make it work

Sunspark-007, 2021-10-07:
The connector has been located and confirmed working on a 1995 Kinesis Essential with the new controller. https://www.molex.com/molex/products/part-detail/ffc_fpc_connectors/0520441345

https://www.molex.com/pdm_docs/sd/520441345_sd.pdf

It's listed as obsolete, but there is still inventory left.

aleb, 2022-03-14:
Below are the pictures of a KB133 and some distances I measured. The board is symmetric from what I can tell. 

The shape of the side-holes is different on this board (?), but hopefully it fits.

Looking at the side with four LEDs, the right-side connector has to be moved down (towards the LEDs) 4.5 mm and left (towards the center) 3.7 mm. Similarly for the left-side connector.

Would such a change work for KB500+?

I see it's easy to move the connector in the KiCad PCB editor, but the traces are not moved automatically. Any suggestions how to move them, also considering they are on both sides of the board?

[image]

[image]

[image]

stapelberg, 2022-03-15:
Thanks for sharing the details!

> Looking at the side with four LEDs, the right-side connector has to be moved down (towards the LEDs) 4.5 mm and left (towards the center) 3.7 mm. Similarly for the left-side connector.
> 
> Would such a change work for KB500+?

While it is **possible**, I don’t think such a change would be ideal:

1. The location of the connectors has been chosen to match the original PCB’s location as close as possible. This ensures our board doesn’t do any worse in terms of putting strain on the flex cables.
2. The further down we move the connectors, the less space you will have between connectors, which makes them really inconvenient to open/close without resorting to tooling and/or damaging your board accidentally.

Given how rare the older Kinesis are, maybe we shouldn’t bother with changing the main kinT design to work for them, and instead just keep a variant around.

> I see it's easy to move the connector in the KiCad PCB editor, but the traces are not moved automatically. Any suggestions how to move them, also considering they are on both sides of the board?

If you select the traces and the connector with your mouse (drag open a rectangular selection) and then press `m` to move, kicad will move both. I think some cutting/re-connecting will be required, but ideally in the part where the traces are straight, so that it’s easy to do.

aleb, 2022-09-07:
https://aleb.ro/post/2022-06-01-refurbishing-kinesis-KB133/

stapelberg, 2022-09-08:
Thanks for sharing this!

pentuin, 2022-09-09:
I'm close to giving up. I have a KB500 but it has those rotten stinkin' PCB connectors instead of the flex cable. I installed Millmax sockets on the keywell pcbs but none of the keys in the left keywell work and only 17 keys in the right  keywell. I don't think the sockets are the problem, though.

BTW, all of the function keys and the thumb keys work fine (and there are sockets in the thumb keys too, of course. Also the same keys that don't work with the Stapleberg controller also don't work with the original controller.

I've been doing a lot of continuity testing and checking the diodes and most of the key switches work and the passthrough from the anode to the key switch works  on all but one of the keys. The continuity from the pcb connector to the keys and test connections are terrible. Almost none of them work. Frankly I'm not sure whether they're supposed to work, but the traces don't look damaged or anything.

It seems to me that except for about 5 keys the problem is entirely in the pcb connector, so I'm wondering if there is anything else I can try. 

1. I could get some wrapping wire and bridge the dead pcb connector traces to a trace terminal that still works. (Pardon me, but I don't know the correct nomenclature.) 
2. I could remove all of the MillMax sockets and permanently solder the switches, though as I said I doubt that's the problem.
3. I could try manually wiring the keywells, beyond just the bridging. in other words just throw away the pcb and start over.
4. Something else?

BTW, where do you get that pcb editory Aleb is using? What's that?

Here are a few pictures:

[image]
[image]
[image]
[image]

aleb, 2022-09-09:
You can install [KiCad](https://www.kicad.org/), `git clone https://github.com/kinx-project/kint.git` and then open `kint/kint.kicad_pro` in KiCad.

Read my blog post for how I checked the connectivity between the Teensy pins and the paths on the thin semi-rigid PCB.

I see you soldered straight pins into J7 and J2. Will it fit when closing the case?

pentuin, 2022-09-09:
> I see you soldered straight pins into J7 and J2. Will it fit when closing the case?

Yes. Same height as the pins holding the teensy. Will check out KiCad, thanks!

pentuin, 2022-09-10:
Where do you get the files for the Kint project, to open in KiCad? Note that my version is from June 2020. Do I need the specific PCB file for that version?

aleb, 2022-09-10:
You can install [KiCad](https://www.kicad.org/), `git clone https://github.com/kinx-project/kint.git` to get the project on your computer and then open the `kint/kint/kint.kicad_pro` file in KiCad. 

This will get you the latest version, I think the paths on the board have not changed so you should be able to use it.

pentuin, 2022-09-11:
Thanks! Here's a pdf I just created if you want to use dye sublimation to create legends on a set of blank pbt keycaps. The primary legends are Dvorak, and the secondary QWERTY. I also used my own designs for a few of the keys, like the CNTRL and ALT keys and the Enter and Space keys, Caps Lock, and some others. (The ALT Key is the logo for the Thunderbolts Project, because it's Alternative Physics. :) The keys Kinesis uses are basically OEM profile keys, but you could use Cherry too, if you can find blanks in a light color. Most of the standard full keycap sets have enough 1.25U keys, but they're all probably row 4 so you might need to pick up individiual 1.25U keys for rows 1 thru 3 or just use some standard XDA or DSA.
[Kinesis Key Legends.pdf](https://github.com/kinx-project/kint/files/9542048/Kinesis.Key.Legends.pdf)

pentuin, 2022-09-14:
I clearly don't understand how these molex connectors are supposed to work. The connection to the traces on the pcb are made by *pressure*, but there's nothing to exert enough pressure on the pcb connectors for them to make a connection. I don't understand how this is supposed to happen, or why it worked in the past but doesn't work now. Very frustrating. The pressure on the connection that activates it is *mechanical*, but the little plastic "wedge" inside the molex are on the wrong side. But they are obviously making some connection, it's just not at all robust and very flakey. You put the pcb connector into the female receptacle in the molex and then place the probes at the contact points on the pcb and there's no connection to the molex leads until you push down on them. I've been figuring that the "shroud" or "jacket" with that flat piece that fits inside the receptacle was there to create the pressure, but if so, it doesn't work. And it doesn't work on any of the molex connectors I've tried. I got some brand new ones, and you still have to push on the pin from the outside to get a connection!

How strange! How in the world is this supposed to work?

At this point I've done so much plugging and unplugging that the PCB is pretty much wrecked, so I'm going to have to hand wire the thing... and I'm NOT going to use the "molex" connectors. It's almost as if they're designed NOT to work, or something. (I know it's called an FCC or something but I can never remember what it's called.)

aleb, 2022-09-14:
Do you have a photo of the wrecked part?

pentuin, 2022-09-15:
Aleb:
I'm not sure what is wrecked about it. Certainly some of the pin connections look distorted because of the number of times I've pulled the switches out and the difficulty of getting them back in. This is because I tried to use the Millmax sockets. At this point it was a bad idea, because it's almost more trouble than just soldering the switch pins, although it's a close call.

But the real problem is that nothing works. I've checked the continuity and done some repairs of damaged leads/traces so all of the switches and diodes check out. But nothing on the board works, so there's something going on with this flex pcb connector and I can't figure out what it is. If I press down on the place where the connector "fingers" touch the traces I can get continuity, but I can't figure out why this is. It seems like the fingers should not require that to achieve a connection. If there's some king of break or crack there, it's not visible. I've tried three different FFC/FBC connectors and none of them achieve continuity. There is clearly something exotic going on with the PCB "connector" (the 'ribbon' from the PCB), but I have no idea what it is. It doesn't seem logical that there should be discontinuity between the "fingers" and the traces when there is no visible damage whatsoever. Anyway, here's a couple of pictures of the left hand key well pcb. Don't hate me because I'm beautiful :) 

[image]

[image]

pentuin, 2022-09-15:
I have ordered some new diodes and some single strand wrapping wire so I can just directly wire the switches. Anyway, here's a picture of a diagram I made after continuity testing of the two PCBs. I labeled the pins to conform to the way they're labeled on the KINT controller, where the top row for the left begins with '1' and the top row for the right begins with '13' for reasons that may make sense to someone. The notes in the lower left indicate where the discontinuities where, which I have repaired with patch wiring to no avail.
[image]

pentuin, 2022-09-18:
I was all set to hand wire my board and then stepped on and broke my soldering iron. Had to buy a new one so I'm waiting for it. I still don't understand how this PC connector works though. The whole this seems tres odd.

pentuin, 2022-09-20:
Well, I have concluded that the stress placed on the FPC strips because of the ways I had to modify the FFC molex connections broke or cracked the connection between the fingers and the traces on the PCBs. This explains why pressing on that joint re-establishes continuity as long as I continue the pressure. I don't think there is any way to come back from this, so I've given up on the keywell PCBs as unsalvageable. I plan to manually wire the the switches, and have worked out the routes for that. There are 5 diode routes and 6 key switch routes for each of the two keywells. It shouldn't be difficult to use single strand wire for the key switch connections.

aleb, 2022-09-20:
Maybe you can save the terminal paths on the FPC strips by adding some solder to each, to make them thicker.

pentuin, 2022-10-29:
I gave up on the flimsy PCB and have just hard wired. I'm getting some flakey results such as two characters getting output instead of one, but at least most of the keys get some result. I've tested continuity and because of the size of the wrapping wire I used (I think) there is one column that seems to have a dead connection. I'll post some pictures, but it's not pretty.

pentuin, 2022-10-30:
I finally have most of the keys working except for the ?/,Z key and the insert key. As far as I can tell the insert key doesn't do anything anyway, but I do occasionally need the Z key.

Also, can I still use QMK to program my keyboard with Teensy 4.1? I can't make heads or tails of the instructions, as they don't tell you what the program is actually doing, what files it's creating (for instance, where are all the default files)? Did I make a huge mistake getting the 4.1 chip? Is the configurator ever going to work with that chip?

stapelberg, 2022-10-30:
Yes, you can use QMK for the kint41 (since QMK 0.14.0, current version is 0.18). I’ve been using the kint41 (with Teensy 4.1) for years and it works fine.

Follow the tutorial at https://docs.qmk.fm/#/newbs

Once you have a working build environment, I use a command like `make kinesis/kint41:stapelberg:flash` to build and flash QMK. You might want to substitute `make kinesis/kint41:default:flash` until you define your own keyboard layout.

pentuin, 2022-10-30:
Thanks, they helps. BTW, the PRGM key seems to take the keyboard offline. Is that some kind of reset?

stapelberg, 2022-10-30:
Yes, it resets into the bootloader (for updating your firmware). See https://github.com/qmk/qmk_firmware/pull/18802/commits/57045a2ce929ea02cf4b2917ad8001816c41e41b if you want to move it out of the way.

pentuin, 2022-10-30:
Ah thanks. I thought it was something like that... so you don't have to open the keyboard to flash a new firmware. Great!

pentuin, 2022-10-31:
Still can't get the thsee key to work, and the shift keys seem to TURN ON CAPS LOCK until I turn it off. The delete key also doesn't work. The rest of the keys seem to work fine, although since I'm using the QWERTY default with the Windows Dvorak some of the side keys are jumbled up, like the [{, ]}, |\, -_, ?/, and =+. Those are understandable. 

Now the shift keys work, again. Weird. SO NOW THEY DON'T. WEIRD. Using the shift keys with the character keys instead of alphabetic keys, seems to reset the shift keys back to normal. Double Weird.

I have checked the delete and thsee keys' electronics numerous times and they all seem OK so I'm thinking the problem is either with the board or the firmware. If I can figure out how to use the KiCad program to check the circuits that should advance things a bit. Anyway, it looks promising, and I'm using the Kinesis to type this, so it more or less works again.

pentuin, 2022-10-31:
Oh, one other thing. When I first plug in the board and hit the thsee key I get a string of repeating thsees until I press some other key which stops the repeat. From then ono the thsee key doesn't work.

pentuin, 2022-10-31:
OK, here are a couple of pictures from KiCad. The connection to pin 7 is for the diode row that connects to the tshee key (pronounced like the first letter is tshee-oo where they keep the animals). The connection for pin 13 is for the column connection. I don't have the experience to determine precisely what's wrong, so thought someone could give me a hint.
[image]
[image]

stapelberg, 2022-10-31:
You can also use `debug_config.matrix = true;` setting to easily find out which row/column a key belongs to when you press it: https://michael.stapelberg.ch/posts/2021-05-08-keyboard-input-latency-qmk-kinesis/#finding-the-pins

Your symptom sounds like a faulty soldering connection. Locate the affected row(s)/column(s) and double-check/re-do their soldering.

pentuin, 2022-10-31:
> Your symptom sounds like a faulty soldering connection. Locate the affected row(s)/column(s) and double-check/re-do their soldering.

I've done that three times now. I'm pretty sure the problem isn't with my soldering of the keyboard. Rather, it might be a short on the controller. I had to desolder the connections for the Molex connectors so will check that again. But it just seems weird that it's the "Dvorak Z" key that's the problem, which is the "?/" key in the QWERTY layout.

I also check to see that the signal was being sent to the right pin when the "Z" key is pressed, and it is. That suggests, again that the problem is on the controller board. The same is true with the Delete key. The correct signal is going to the pin.

pentuin, 2022-10-31:
OK, I finally tracked it down. The bad delete key was easy, because I was using hotswap sockets for the thumb switches, so just remelted and added a little solder. The clue to the problem with the "z" key was the fact that whenever I first plugged in the keyboard I'd get a series of z's (ZZZZZZZZZZZZZZZ). Apparently when I soldered the column the silicone insulation on the wire softened up enough that the diode lead melted through and shorted it. So I just put some shrink tubing over those sections and resoldered them, and I can now type the "z" as much as a like. The only problem left is that the keypad doesn't work. Since I never touched those it must be the molex connection on the controller that's poorly soldered and I'll check that later when I figure out which pin on the molex is the right one.

So now I'm ready to program this thing!

aleb, 2022-10-31:
The Keypad key is mapped to `KC_NO` which is to nothing: https://github.com/qmk/qmk_firmware/blob/master/keyboards/kinesis/keymaps/default/keymap.c

pentuin, 2022-11-01:
Success! It works! I even defined the numeric keypad layout and got the toggle to work. Back in business!!

Thanks a lot, guys. Took me a couple of months but it was worth it. I was never able to flash the firmware I created using the command lines or the toolbox so just used the Teensy Loader. The cherry mx blue switches are also a big improvement over the browns. I'll post a picture later of the rats nest I hard wired, later. It's pretty nasty looking, but for posterity...

stapelberg, 2022-11-01:
Glad to hear you could get it to work now! :)

masters3d, 2022-11-06:
As an alternative, if the keyboard still working fine and if you have the version that is PS/2, you may want to try a PS2 to USB with QMK. I just brought one but it will take a few weeks to get https://www.tinkerboy.xyz/product/tinkerboy-ps-2-keyboard-to-usb-with-soarers-converter/

mezzopian0, 2022-11-06:
Wow, thanks, this thread (and the kinT) might possibly save my keyboard.

I have a very old (25 years) KB133PC with AT connector that seems to be stuck in a strange mode (scrambled up). Essentially, the right key well works normally, while the left one sends key codes that do not make sense to me ("A" sends shift, most keys send numeric keypad codes, some click but do not send any code). The thumb clusters are also messed up (they send divide, decimal key, print screen, etc.) and some keys do not send any code. The NumLock LED is lit.

I cannot reprogram the keyboard because the top row of rubber keys (Esc, Fn, ... Keypad, Program) is no longer working (they do not send any key code and no click sound is played back when I press them).
All the hard plastic keys (the two key wells and thumb clusters) play click sound when I press them, so I suppose they are recognised by the controller.

I was planning to use the keyboard again. I don't need the row of FN keys, so that's not a big issue: I'd just need to remap Esc to some other key. 
I had considering using a PS/2 to USB converter to fix the mapping, like tinkerBOY w/ QMK (I think I bought a converter with Soarer some years ago but never got to try it): the issue is that, since not all keys send a KeyCode (a click is played by the keyboard, but no key code sent to the PC), as far as I understand, I would not be able to use all keys.

Is there anything that can be done to "reset" the keyboard to factory settings or at least to a state where each key sends a code?
If that's not possible, I suppose I could de-solder the cables/connectors from the Kinesis board and replace it with a kinT controller. There is much valuable information in here, thanks stapelberg, aleb, and pentuin!

mezzopian0, 2022-11-06:
This is how my board looks like.
[image]

aleb, 2022-11-06:
Fixing the functions keys rows could be easy with a pencil (graphite) and glue. Check out https://aleb.ro/post/2022-09-14-fixing-kinesis-function-keys/

pentuin, 2022-11-11:
I'm getting a loot of letter bounces. Is there a way to adjust this in QMK?

stapelberg, 2022-11-11:
> I'm getting a loot of letter bounces. Is there a way to adjust this in QMK?

Yes, increase the `DEBOUNCE` definition in `keyboards/kinesis/kint41/config.h`. See also https://docs.qmk.fm/#/feature_debounce_type

dm17, 2024-10-24:
Curious if anyone has tested instructions for acheiving this on the ps/2 kinesis yet.

aleb, 2024-10-25:
Yes read the thread above. https://aleb.ro/post/2022-09-08-refurbishing-kinesis-KB133/