kinx-project/kint #45: Firmware flashing issue with Kint36

iparnelly, 2021-12-11:
Hi Michael,

First off, thanks so much for this project - as a beginner it has been a lot of fun to experiment and mess around with.

I recently compiled some changes to my configuration using the QMK configurator, but when I attempt to flash the firmware onto my keyboard my keyboard becomes unresponsive. The orange LED on the teensy does not light up, and the 4 blue keyboard LEDs stay permanently lit. The teensy loader does not seem to indicate any issue (neither does the verbose logging). 

I suspect that this is an issue with the QMK software and how it is compiling the firmware, but to be honest, I'm not sure where to begin in diagnosing this problem. It's worth mentioning that I have tested this using the web UI for QMK configurator, as well as the qmk cli tool on both Mac and Windows (same behavior across interface/platform).

For what it's worth, the pre-built tested firmware that you have linked on the readme [here](https://github.com/kinx-project/kint#installing-the-firmware) works fine, but the QMK Configurator [link](https://config.qmk.fm/#/kinesis/kint36/LAYOUT) you have listed in the same table does not work for me anymore (it did previously).

I'm happy to create an issue on the QMK repo, but I'd appreciate some insight or direction on how I can dive in to this more.

Thanks!

stapelberg, 2021-12-11:
Wild shot into the blue: the only major recent change was EEPROM-related. In your QMK build, is https://github.com/qmk/qmk_firmware/pull/12947/files included? If so, I wonder if reverting it helps.

If yes, you might need to clear the EEPROM on your Teensy.

iparnelly, 2021-12-12:
That commit was included in my QMK build. Once I reverted that commit out then I could successfully flash newly compiled firmware. I'll look into clearing the EEPROM tomorrow and testing that with the updated QMK build, thanks for the advice!

kpence, 2021-12-17:
> I have tested this using the web UI for QMK configurator, as well as the qmk cli tool on both Mac and Windows (same behavior across interface/platform).

> For what it's worth, the pre-built tested firmware that you have linked on the readme here works fine, but the QMK Configurator link you have listed in the same table does not work for me anymore (it did previously).

It's the same for me.

kpence, 2021-12-17:
Reverting that commit fixed the issue for me, thanks!

zemackdaddy, 2021-12-21:
@stapelberg I too am having this exact same issue.

I don't really know how to revert the commit as it compilation fails due to many errors and I'd rather have an official fix.

I did hold in the reset button for like 20 seconds because I read that it will reset the EPROM and that did not fix the issue.

I raised an issue here but any advice you might give would be great.

https://github.com/qmk/qmk_firmware/issues/15521