kinx-project/kint #46: Teensy 4.1 and USB

numerys, 2021-12-30:
I want to build a split keyboard which daisy-chains (eg. computer---kb_left---kb_right---mouse) and  I would like to use mainline-qmk (python3 -m pip install --user qmk).

For that I would like to use 2 x Teensy 4.1. For each side one Teensy. Is it possible to use the internal usb-port of T4.1 with kint? That would be my main-concern. I guess the compilation / installation will be ok, but what are your thoughts about the usb-port of the T4.1? Is there any chance to daisy-chain?

brandondees, 2021-12-31:
I'm also curious about this possibility (any controller, just being able to do USB-based chaining at all, as opposed to TRRS etc, or more preferably a way to act like a usb hub with one or more arbitrary devices plugged in). I don't mind doing a little programming but I'm a little out of my element here and want to avoid wasting time on unrealistic ideas.

stapelberg, 2021-12-31:
> For that I would like to use 2 x Teensy 4.1. For each side one Teensy. Is it possible to use the internal usb-port of T4.1 with kint? That would be my main-concern. I guess the compilation / [...]

kint is just a hardware board, so I think your question should read “is it possible to use the second USB port on the Teensy 4.1 with QMK” (because you’ll likely use your own board).

QMK uses ChibiOS, which in turn uses [NXP‘s SDK](https://github.com/NXPmicro/mcux-sdk). While the NXP SDK supports both USB ports, ChibiOS doesn’t, and QMK doesn’t either.

Adding the required support requires decent familiarity with how USB stacks work (and how NXP’s works in particular), and even then will take weeks to months.

Hence, I don’t think it’s a good idea to use USB for connecting the two controllers.

My recommendation would be to use a different way of connecting the two controllers. What speaks against TRRS?

brandondees, 2021-12-31:
that's a helpful answer from my perspective, thanks!

the use i had in mind was to have things like a yubikey or fingerprint reader mounted on the keyboard.

stapelberg, 2021-12-31:
For that, I recommend using a USB hub.

As it happens, I have published a design that you can build: https://michael.stapelberg.ch/posts/2018-04-17-kinx-usb-hub/

brandondees, 2022-01-01:
ooh \_very\_ helpful!

\-------- Original Message --------

[view it on GitHub]: https://github.com/kinx-project/kint/issues/46#issuecomment-1003461522
[unsubscribe]: https://github.com/
[AABNLBF4GIBBBJTNPGSZZ7DUTYW7JA5CNFSM5K7VA5A2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHPHZXEQ.gif]: https://github.com/notifications/beacon/AABNLBF4GIBBBJTNPGSZZ7DUTYW7JA5CNFSM5K7VA5A2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHPHZXEQ.gif