kinx-project/kint #49: Need some help debugging

fuleow, 2022-03-01:
I sourced some PCBs from JLC PCB using the zip file in this repository and built the kint today for my KB500 using the Teensy 4.1. I pretested and preflashed the Teensy before assembling the rest of the board. I soldered the LEDs first and pretested again and when I toggled capslock, numlock, scroll lock on my other keyboard the lights on the kint turned on and off so it appeared to be working.

Unfortunately after I had everything re-assembled it did not work. Pressing a key on the left keywell (e.g. Q) would cause random modifier keys to be pressed. I thought it might be an issue with the thumb cluster since I had to hand rewire those for the KB500, so I disconnected it but the issue still persisted. I tried only connecting the top row function keys and qmk tester register the key presses but also registers a bunch of LCTRL, LALT and LOS. 

I visually inspected the board and didn't see any issues with soldering but probing with the multimeter I saw that pin 6 on the teensy was shorted with ground. I'm pretty sure this should not be the case, but I am not sure how this is possible. There doesn't appear to be any ground pins near any of the pin 6 connections.

Since col 6 is related to the left thumb cluster this makes me suspect it might be the issue. Do you have any suggestions or tips on how I should try to debug the issue further? Thanks.

stapelberg, 2022-03-01:
My first suggestion would be to disassemble everything again and then re-assemble it step by step: connect only one key block. Then connect the next, etc. That way, you should be able to figure out where the problem is located, and then go from there.

Do you get the connection to GND also when nothing is connected to the board?

fuleow, 2022-03-01:
Yeah I tried disconnecting everything and only connecting the right function row (F9 - Pause). The keys register, but I get random key presses after awhile. Yes, I do get connection between GND and Pin 6 with nothing connected. I tested the 4 other unpopulated boards from JLC and they don't have this issue so it doesn't appear to be a PCB problem

```
KEY-UP   - QMK: KC_PSCR Event key: PrintScreen Code: PrintScreen   KeyCode: 44 in NaNms
KEY-DOWN - QMK: KC_F12  Event key: F12         Code: F12           KeyCode: 123
KEY-UP   - QMK: KC_F12  Event key: F12         Code: F12           KeyCode: 123 in 165.000ms
KEY-DOWN - QMK: KC_F11  Event key: F11         Code: F11           KeyCode: 122
KEY-UP   - QMK: KC_F11  Event key: F11         Code: F11           KeyCode: 122 in 190.000ms
KEY-DOWN - QMK: KC_F10  Event key: F10         Code: F10           KeyCode: 121
KEY-UP   - QMK: KC_F10  Event key: F10         Code: F10           KeyCode: 121 in 183.000ms
KEY-DOWN - QMK: KC_F9   Event key: F9          Code: F9            KeyCode: 120
KEY-UP   - QMK: KC_F9   Event key: F9          Code: F9            KeyCode: 120 in 242.000ms
KEY-UP   - QMK: KC_LCTL Event key: Control     Code: ControlLeft   KeyCode: 17 in 7996.000ms
KEY-UP   - QMK: KC_RGUI Event key: OS          Code: OSRight       KeyCode: 91 in 7988.000ms
KEY-UP   - QMK: KC_BSPC Event key: Backspace   Code: Backspace     KeyCode: 8 in NaNms
KEY-UP   - QMK: KC_DEL  Event key: Delete      Code: Delete        KeyCode: 46 in 7983.000ms
KEY-UP   - QMK: KC_PGDN Event key: PageDown    Code: PageDown      KeyCode: 34 in NaNms
KEY-UP   - QMK: KC_PGUP Event key: PageUp      Code: PageUp        KeyCode: 33 in NaNms
```

fuleow, 2022-03-01:
I've taken some pictures if it helps: https://imgur.com/a/fW0qaJa

fuleow, 2022-03-02:
Decided to purchase a prebuilt instead so I’m closing this issue

closed 2022-03-02 by fuleow

fuleow, 2022-03-02:
> It’s not a PCB layout problem, but it might be that one of the PCBs you got from JLC has an issue.

Yes, sorry I did not mean to imply the gerber files on this repo were incorrect. I thought to check the others as a sanity test, thinking that if a manufacturing defect was the cause it would show up on the others since it's the same batch.

> One quick thing to try, if you have given up on this particular build anyway: cut the pin 6 trace with a scalpel close to the teensy, then connect it to the molex connector with a jumper wire. That should circumvent the PCB and any faulty traces entirely.

Great suggestion! I cut the trace near pin 6 on the teensy and pin 6 was no longer connected to GND. The COL_6 pins on J1 and J5 were still correctly connected to each other and they were still connected to GND. I also checked GND against every other pin on J1 and J5 and there was no connection.

At this point I'm inclined to think it's a PCB manufacturing issue. My prior experience with flux residue or inadvertent bridges is there's an undesirable connection with an adjacent pin. If my understanding of the kint layout is correct there shouldn't be any through holes on the keywell connectors that are directly connected to ground.

> You could try soldering one of the other ones.

This was my thought as well, but I was not able to desolder the teensy successfully. I think if I had to do it over I would opt for female headers on the board so the teensy could be salvaged in case I made a mistake. The cost of trying again would be much lower if I only needed to re-purchase the connectors.

Maybe I'll give it another go in the future if I acquire another kinesis. Perhaps a teensy 4.0 with headers and female headers on the kint board. That would reduce the number of soldering required and allow the teensy to be easily disconnected.

Assuming the capslock LED is optional I would omit that too and consequently the vertical header for the teensy.

stapelberg, 2022-03-02:
> This was my thought as well, but I was not able to desolder the teensy successfully

Oh, yeah, desoldering is essentially impossible. 

I can recommend using these (or similar):

https://www.digikey.ch/short/377hm238

https://www.digikey.ch/short/5b0rpwnq

They are small enough that you can still close the keyboard, and you can swap out the teensy as you wish.

> Maybe I'll give it another go in the future if I acquire another kinesis. Perhaps a teensy 4.0 with headers and female headers on the kint board. That would reduce the number of soldering required and allow the teensy to be easily disconnected.

Using a traditional male/female pin header might result in a too bulky controller. There is not a lot of space in that part of the keyboard.

> Assuming the capslock LED is optional I would omit that too and consequently the vertical header for the teensy.

Yeah, for a minimal build you can go without LEDs.

fuleow, 2022-03-02:
Thanks for your time with the advice and help @stapelberg ! I also enjoyed reading your blog, and ordered a Logitech MX Ergo because of it.