kinx-project/kint #50: 1. Built-in speaker sorely missed 2. Extra buttons

Megavvolt, 2022-04-14:
Not an issue as such! Just a couple of questions!
If they are inappropriate here please tell where else to look for answers.

1. Is there a way/are there plans to teach **kint** make "clicks" on key presses? I noticed when the sound was off I almost never bottomed out. I miss that.
2. One of my Kinesis Contoured keyboards had 5 extra buttons (I simply drilled holes and added buttons). Is it possible to pull the same trick here?

stapelberg, 2022-04-15:
> * Is there a way/are there plans to teach **kint** make "clicks" on key presses? I noticed when the sound was off I almost never bottomed out. I miss that.

The QMK firmware itself supports the “Audio Click” feature: https://docs.qmk.fm/#/feature_audio?id=audio-click

However, the required drivers for each hardware platform are not necessarily implemented, so whether you can use this feature depends on which Teensy microcontroller you use. For example, support for the Teensy 3.6 is tracked in  https://github.com/kinx-project/kint/issues/36. I think the only Teensy with which this feature currently works is the Teensy++ 2.0, but I haven’t checked if somebody might have contributed changes to QMK since I last looked, so best do your own research on this :) 

It’s not a feature I use, so there are no plans from my end to implement this.

> 2\. One of my Kinesis Contoured keyboards had 5 extra buttons (I simply drilled holes and added buttons). Is it possible to pull the same trick here?

Sure, if you wire them up to the Teensy controller and fit them into the key matrix, that should work.

closed 2022-04-15 by stapelberg

Megavvolt, 2022-06-13:
@aleb not sure if it's ok to spam here but here's a link to a great forum where you'll find almost everything you might want to know on adding buttons:
https://geekhack.org/index.php?topic=26579.msg2252512#msg2252512