kinx-project/kint #52: Teensy 4.1 has stopped working properly?

JamyGolden, 2022-04-25:
## Description

I put together a Teensy 4.1 kinT keyboard controller a few months ago and it has been working amazingly up until this morning. The keys don't all work correctly any more, some seem to work for a while and then just stop, sometimes random strings are outputted and sometimes space or a key is held so it outputs something like "tttttttttttt" continuously. It seems to have gotten worse over the course of the day in the sense that no key presses seem to work correctly any more.

I last flashed the firmware about 2 months ago and I don't move the keyboard around, so there hasn't been any physical knocks or anything.

## Debugging

1. I flashed the board hoping the qmk firmware somehow degraded over night, that didn't work.
2. I flashed with the standard stapelberg keymap, thinking maybe my map was somehow corrupt and qmk wasn't picking it up, that didn't work.
3. I made sure the physical connections on the board were all tight and nothing had somehow become a bit loose. Everything was fitted correctly and nothing changed when I tried to use the board again.

I'm not sure what else to try? Have you heard of an issue like this before? Do you have any suggestions?

stapelberg, 2022-04-25:
Is there any pattern to the keys that don’t work?

My wild guess would be that one the keyboard matrix column lines (`COL_*` in the schematic) has become defective (cold solder joint). My recommendation would be to re-heat all the solder joints on the PCB, possibly starting with the column ones and then checking if that fixed the issue.

I haven’t heard of such an issue before.

In the past, when a Teensy stopped working for me, it would typically stop working entirely. I.e., when turning on, the microcontroller would get hot, and never enumerate on USB. When only individual keys don’t work, my guess is a bad connection.

JamyGolden, 2022-04-25:
Yeah definitely a strange issue. The patterns tend to change a bit each time it's unplugged and re-plugged in, but it could be like all keys don't work except the windows key acts like alt, or all keys work except qwerty row.

Thanks for the suggestion, I'll make sure everything is soldered correctly and let you know how it went.

JamyGolden, 2022-05-25:
I re-soldered some connections. While I wasn't able to see any issues, it's working again :tada: :keyboard: . Hopefully permanently this time. Thanks for the suggestion.

closed 2022-05-25 by JamyGolden