kinx-project/kint #55: Teensy LC pins 24, 25 for controlling LEDs

aleb, 2022-06-01:
I see the Teensy-LC has pin 24 and 25 which seem unused. Can these be routed to control two more leds?

[image]

aleb, 2022-06-03:
I tried on a breadboard and pin 24 is usable, but I did not manage to make pin 25 working.

I see pin 24 is assigned to the "keypad" led according to the PCB: 
[image]

But in [config.h](https://github.com/qmk/qmk_firmware/blob/83a32b257473c5dc9fbacd71a3c1470dde21ad84/keyboards/kinesis/kint41/config.h#L101) it's assigned to the "compose" led:
```
#define LED_PIN_ON_STATE 0
#define LED_NUM_LOCK_PIN LINE_PIN26
#define LED_CAPS_LOCK_PIN LINE_PIN12
#define LED_SCROLL_LOCK_PIN LINE_PIN25
#define LED_COMPOSE_PIN LINE_PIN24
```

I can't even make the scroll-lock led turn on on the keyboard I built with Teensy 2.0++ and I have absolutely no clue where the Num-pad key is. So given only the caps-lock LED is useful IMO, this effort would be pointless.

A new pin 24 could be easily added and routed to its doppleganger for a quick win, I guess, if anybody needs this "compose" led, whatever it might be. Pin 25 might require some debugging. I'll make a QMK PR to add support for Teensy LC once I integrate it in a KB500 and check it works.

aleb, 2022-06-03:
See https://github.com/qmk/qmk_firmware/pull/17301 for Teensy LC support. At the moment I only tested that the Caps-lock and Scroll-lock LEDs are updated. To be tested once I integrate it into a KB500.

stapelberg, 2022-06-04:
Thanks for taking a look at this and sending the QMK pull request!

I found some documentation about the different LEDs at https://github.com/qmk/qmk_firmware/blob/master/docs/feature_led_indicators.md

It seems like the compose LED is entirely unused within QMK by default. I only wired it up so that users have the option of using the LED for their own custom purpose, if they need it :)

sigprof, 2022-06-04:
The compose LED support in QMK works… if you actually find a way to control it from the host side. I'm not sure whether it is possible to control it from Windows at all (there might be a way to do something with the Kana LED though, but nothing that looks like Compose). And recent Linux distros use the `libinput` driver for HID keyboards, which does not have support for any indicators outside the standard Caps/Num/Scroll Lock set. The `evdev` driver can control the Compose and Kana indicators, but there is a mismatch in the indicator numbering between the driver and the corresponding description in the `xkeyboard-config` package; I'm not sure whether there is any point to open a bug report for that though, because the `evdev` driver is deprecated in favor of `libinput`.

aleb, 2022-06-08:
The Teensy (LC) LED light is quite visible when the room is dark, as the orange light reflects on keys `5` and `T`. Is this also an issue for Teensy 3.6 or 4.1? I see we do make an effort in [`kint41.c`](https://github.com/qmk/qmk_firmware/blob/816accd/keyboards/kinesis/kint41/kint41.c#L24) etc to turn it on.

stapelberg, 2022-06-08:
It’s not as pronounced in my teensy 4.1 build:

https://user-images.githubusercontent.com/55506/172718612-94445b54-6213-46b7-8d80-29be3550e178.MOV