kinx-project/kint #57: Teensy 3.6 pin mapping

basketofkittens, 2022-07-13:
I'm doing a custom keyboard and I'd like to use QMK with a Teensy 3.6. This config seems to be real close to what I need, but I'm a little confused on the pin mapping.

`/kinesis/kint36/config.h` has the following lines:
```
#define MATRIX_ROW_PINS { D3, C3, C4, C6, D2, B0, D7, A12, A13, B17, B16, D0, B1, C2, D6 }
#define MATRIX_COL_PINS { B3, D1, C0, D5, C1, B2, D4 }
```

These don't really make any sense to me in the context of a Teensy 3.6 or with the pins that are wired out on the kinT board itself (eg ROW_0 is tied to Teensy pin 0, which for a 3.6 is PTB16). 

Sorry if this is more of a QMK-specific question, but I'm new to all this and it's proving to be an absolute jungle.

stapelberg, 2022-07-13:
Have a look at https://github.com/qmk/qmk_firmware/issues/13052 for some context. QMK started out on AVR microcontrollers, that’s where the names come from.

basketofkittens, 2022-07-17:
Thanks for the quick reply! Sorry I couldn't get back to this sooner. 

It seems I had made an opening assumption while looking at the schematic that the ROW_0 net on the schematic was tied to the 0th entry of the MATRIX_ROW_PINS list. This is what screwed me up. I thought there was some kind of funky re-mapping going on between the MK6x port/pin naming and the config. Upon a closer inspection it's clear that the config is a simple 1:1 mapping of the MK6x port/pin numbers. This makes it nice and easy. 

Thanks again for the help!

closed 2022-07-17 by basketofkittens