kinx-project/kint #58: Is there any technical advantage between 3.6 and 4.1?

rockybulwinkle, 2022-08-08:
As title says, I'm wondering if there is any technical reason to pick 3.6 vs 4.1 in terms of performance or functionality?

I want to make my keyboard wireless and battery powered. If there isn't a substantial difference in performance or functionality between the two, I should pick whichever uses less power.

Thanks!

stapelberg, 2022-08-08:
The 4.1 performs better (see its lower input latency, for example), but that won’t really make a measurable difference to the already excellent level of performance of the teensy 3.6.

See https://forum.pjrc.com/threads/60698-Teensy-3-6-vs-4-0-measured-power-draw-at-various-MHz for a power consumption comparison. With default settings, they’re roughly equal in terms of power consumption. The best solution would be to use a Teensy 4.1 and down-clock it, but I haven’t done that yet (and there might not be easy support for it from within QMK).

So, you can probably chose either one. The 4.1 is a little bit cheaper.

rockybulwinkle, 2022-08-09:
Thanks, that's very informative!

closed 2022-08-09 by rockybulwinkle