kinx-project/kint #62: Teensy 4.1: Random crashes | defective?

MrDeerly, 2023-03-14:
Hey! 

First of all thanks for your awesome project! 

I finished my build recently and ran into the following issue:

**Problem description**
My teensy seems to crash after a while (can be hours, or just minutes) and enters the bootloader mode where the small led at the usb port is permanent on. After this the keyboard is not usable anymore, but still listed with hid_lister. I need to replug the cable to get things working again.  This happens with with the firmware you linked in the readme (_QMK 0.17.9_).

**Things i did** 
Initially i thought it might be just an issue with the old firmware, so i went ahead and compiled the default firmware with the latest qmk version (_QMK 0.20.1_). While the compilation itself was successful, i started to have issues flashing the teensy. [QMK mysys](https://msys.qmk.fm/) gave me errors when trying to flash the teensy: `error writing to teensy`. I did some research but couldn't find any helpful information about it which is why i used the standart teensy loader for flashing. Flasing worked flawless, **but now** with the latest qmk version the teensy seem to crash straight away (after seconds). Replugging doesn't help at all in this case.

After the unsuccessful firmware update i went ahead and checked the hardware itself for possible shorts, including measurements.. nothing :/  

**Considerations** 

I have a few ideas what could be wrong, but no idea how to debug it best: 

A) Teensy overheats with the old version after quite some time. With the new version instantly?! 
B) The latest qmk version has a bug and is incompatible with teensy4.1 (https://github.com/kinx-project/kint/issues/45). Am i the first person to discover?
C) The teensy is defective. 

Did you ever heard of those problems or experienced them? Do you have any hint how to debug the issue better, or a wild guess?

Thanks in advance!

stapelberg, 2023-03-14:
I think it’s most likely that your Teensy is defective, in particular if you’re getting errors during flashing. Have you swapped the USB cable already? That would be good to verify, just in case.

I haven’t heard from anyone else regarding such symptoms.

In case it helps, I’m currently running QMK at commit https://github.com/qmk/qmk_firmware/commit/543a863766192eab626492ec4f48641743ffc797 on my main keyboard, so that version should work for sure. 

Good luck debugging this, though I would probably just solder another board to see if it works with a new set of components…

MrDeerly, 2023-03-16:
Thanks for your quick answer... much appreciated!

I did some more testing and found out that actually Windows is causing a lot of issues (what a surprise lol). To be more precise it's probably windows defender. On unix I'm able to flash the teensy without any issues. When using the latest version of qmk I still have those reboots, but not instant as before (which still seems a bit weird). With the version you've linked, it seems to be pretty stable (no crash so far, but maybe I was just lucky).

I will definitely try different versions of qmk till if find the (for me) broken version and identify deprecation. Of course only if it really is a software issue.  

One last question if I may: How would you debug this issue? Enable logs and see what happened before a crash?

Thanks!

stapelberg, 2023-03-18:
> One last question if I may: How would you debug this issue? Enable logs and see what happened before a crash?

Yeah, in general, I would recommend to use as many tools as possible when debugging. You can use the HID based console (https://michael.stapelberg.ch/posts/2021-05-08-keyboard-input-latency-qmk-kinesis/#enable-the-debug-console-in-qmk), or you can connect a USB-to-serial adapter and enable the serial console, but that’s a bit more involved on both the software and hardware side.

lread, 2023-07-16:
FWIW, I got a an error when trying a `qmk flash`, the error looked like so:
```
Programming.error writing to Teensy

gmake[1]: *** [platforms/avr/flash.mk:173: flash] Error 1
Make finished with errors
gmake: *** [Makefile:392: kinesis:lread:flash] Error 1
```

I unplugged my keyboard, and plugged in it an retried. 
Same issue.
I then flashed the kint41 hex file with the teensy loader as described [here](https://github.com/kinx-project/kint#installing-the-firmware).

After that I seem to be able to flash with qmk again.

I'm using a Teensy 4.1 and qmk 1.1.2.