kinx-project/kint #64: GMK Configurator Kint41 creates a bin file and not a hex

Chamels, 2023-03-25:
As the title suggests, the GMK configurator creates a bin file and not a hex. Due to this, you can not properly flash the board.
[image] 
When downloaded it saves the bin file.

Chamels, 2023-03-25:
[kinesis_kint41_kinesis.zip](https://github.com/kinx-project/kint/files/11070426/kinesis_kint41_kinesis.zip)
Here's the bin file if anyone wants to see it.

MrDeerly, 2023-03-28:
I also experienced this. I'd suggest to raise issue here: https://github.com/qmk/qmk_configurator

MrDeerly, 2023-05-03:
Got fixed quite a while ago in here: https://github.com/qmk/qmk_firmware/pull/20295