kinx-project/kint #66: kinT (teensy 4.1) with Vial/Via

JasonhjKim, 2023-05-08:
Hey Stephen,
This is not a really an issue, but just description on my failed attempt.
I was wondering if you have tried loading vial/via on to your qmk code before? I'm currently attempting with your kinT41 code base. Only time I got it to successfully create the hex file is by bumping up the `TRANSIENT_EEPROM_SIZE = 500` for `DYNAMIC_KEYMAP_LAYER_COUNT= 2`. However, once I programme it on to my teensy4.1, it errors out with 7 blinks. According to the documentation https://www.pjrc.com/store/ic_mkl02_t4.html 
```
7 Blinks = ARM JTAG DAP Communication Error
    A communication error was detected, but after correctly detecting (4 blinks) and initializing the ARM JTAG DAP (9 blinks). Programs which crash or have unusual startup behavior can cause a temporary 7 blink error at startup. With a known-good program or blank flash chip, this error may indicate a severe signal quality problem or any of the signal wires shorted to other lines which are initially high impedance but become output after a program is running. But these types of hardware issues are unusual, because hardware trouble for 7 blinks requires first successfully passing tests which would cause 4 or 9 blinks. 7 blinks can also be shown if the processor reboots, either by software write to SCB_AIRCR or due to fault or other error handling, while communication is in progress. If you see 7 blinks, program the flash memory with a simple known-good code before investigating hardware trouble.
```

JasonhjKim, 2023-05-08:
rules.mk
```
VIA_ENABLE = yes
VIAL_ENABLE=yes
VIAL_INSECURE = yes
LTO_ENABLE = yes
QMK_SETTINGS = no
TAP_DANCE_ENABLE = no
KEY_OVERRIDE_ENABLE = no
COMBO_ENABLE = no
EEPROM_DRIVER = transient
```
config.h
```
#pragma once
#define VIAL_KEYBOARD_UID {0x7A, 0x7F, 0xB3, 0x95, 0x6D, 0x43, 0x3B, 0x0E}
#define DYNAMIC_KEYMAP_LAYER_COUNT 0
#define VIAL_KEY_OVERRIDE_ENTRIES 4
#define VIAL_COMBO_ENTRIES 4
#define VIAL_TAP_DANCE_ENTRIES 4
#define TRANSIENT_EEPROM_SIZE 200
```
keymap.c
```
#include QMK_KEYBOARD_H

#define QWERTY 0 // Base qwerty

/****************************************************************************************************
*
* Keymap: Default Layer in Qwerty
*
* ,-------------------------------------------------------------------------------------------------------------------.
* | Esc    |  F1  |  F2  |  F3  |  F4  |  F5  |  F6  |  F8  |  F9  |  F10 |  F12 | PSCR | SLCK | PAUS |  FN0 |  BOOT  |
* |--------+------+------+------+------+------+---------------------------+------+------+------+------+------+--------|
* | =+     |  1!  |  2@  |  3#  |  4$  |  5%  |                           |  6^  |  7&  |  8*  |  9(  |  0)  | -_     |
* |--------+------+------+------+------+------|                           +------+------+------+------+------+--------|
* | Tab    |   Q  |   W  |   E  |   R  |   T  |                           |   Y  |   U  |   I  |   O  |   P  | \|     |
* |--------+------+------+------+------+------|                           |------+------+------+------+------+--------|
* | Caps   |   A  |   S  |   D  |   F  |   G  |                           |   H  |   J  |   K  |   L  |  ;:  | '"     |
* |--------+------+------+------+------+------|                           |------+------+------+------+------+--------|
* | Shift  |   Z  |   X  |   C  |   V  |   B  |                           |   N  |   M  |  ,.  |  .>  |  /?  | Shift  |
* `--------+------+------+------+------+-------                           `------+------+------+------+------+--------'
*          | `~   | INS  | Left | Right|                                         | Up   | Down |  [{  |  ]}  |
*          `---------------------------'                                         `---------------------------'
*                                        ,-------------.         ,-------------.
*                                        | Ctrl | Alt  |         | Gui  | Ctrl |
*                                 ,------|------|------|         |------+------+------.
*                                 |      |      | Home |         | PgUp |      |      |
*                                 | BkSp | Del  |------|         |------|Return| Space|
*                                 |      |      | End  |         | PgDn |      |      |
*                                 `--------------------'         `--------------------'
*/

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
[QWERTY] = LAYOUT(
           KC_DEL, KC_F1  ,KC_F2  ,KC_F3  ,KC_F4  ,KC_F5  ,KC_F6  ,KC_F7  ,KC_F8,
           KC_EQL, KC_1   ,KC_2   ,KC_3   ,KC_4   ,KC_5   ,
           KC_TAB, KC_Q   ,KC_W   ,KC_E   ,KC_R   ,KC_T   ,
           KC_CAPS,KC_A   ,KC_S   ,KC_D   ,KC_F   ,KC_G   ,
           KC_LSFT,KC_Z   ,KC_X   ,KC_C   ,KC_V   ,KC_B   ,
                   KC_GRV ,KC_INS ,KC_LEFT,KC_RGHT,
			   KC_LCTL,KC_LALT,
                                    KC_LGUI,
                           KC_BSPC,KC_ESC ,KC_END ,
    KC_F9  ,KC_F10 ,KC_F11 ,KC_F12 ,KC_PSCR ,KC_SLCK  ,KC_PAUS, MO(1), QK_BOOT,
	KC_6   ,KC_7   ,KC_8   ,KC_9   ,KC_0   ,KC_MINS,
	KC_Y   ,KC_U   ,KC_I   ,KC_O   ,KC_P   ,KC_BSLS,
	KC_H   ,KC_J   ,KC_K   ,KC_L   ,KC_SCLN,KC_QUOT,
	KC_N   ,KC_M   ,KC_COMM,KC_DOT ,KC_SLSH,KC_RSFT,
		KC_UP  ,KC_DOWN,KC_LBRC,KC_RBRC,
           KC_RALT,KC_RCTL,
           KC_PGUP,
           KC_PGDN,KC_ENTER ,KC_SPC
    )
};
```

stapelberg, 2023-05-08:
EEPROM support is a mixed bag currently.

Take a look at https://github.com/kinx-project/kint/issues/8 for the Teensy 3.6, where we never succeeded in making EEPROM support work. It worked on some devices, but others wouldn’t boot — had to revert the commit from QMK.

I don’t know what the Teensy 4.x situation is, but I haven’t done any work on EEPROM support, and I’m not sure anyone else has…

JasonhjKim, 2023-05-08:
I can try making similar changes you made [here](https://github.com/stapelberg/qmk_firmware/commit/2fc2c6fc91f0ea663a6791f3ef437e07981bb5cc) for teensy4.1