kinx-project/kint #67: Recommended PCB version?

lread, 2023-06-29:
Hello! Thanks so much for sharing this very cool project!

I was about to get a kinT keyboard controller PCB printed when I noticed that your link to [Buy PCB|OSHPark] shows v2020-06-30, but if I upload kint.kicad_pcb from this git repo to OSHPark, I see v2022-04-15.

Is the latest the greatest? Or is it better to stick with v2020-06-30?

stapelberg, 2023-06-29:
If you want a known-working revision, v2021-04-25 is the latest revision that I have in my keyboards.

But the latest version should be fine, too (there are no known issues at least).

You can look through https://github.com/kinx-project/kint/commits/main to see what has changed when. I think the only two board changes since then are https://github.com/kinx-project/kint/commit/e50db5470b10f921d979dec410ea861ebf73b23e and https://github.com/kinx-project/kint/commit/afac7593b2d746b0df2951f4b0d2ef72b4ce3b4c. If you don’t want to use a Teensy++, you can ignore them.

lread, 2023-06-29:
Thanks so much for your answer @stapelberg, very helpful!

lread, 2023-07-02:
I'm going to try out the latest and report back my extreme success here.
(to give us one more data point on this newest version).

I guess there is no real reason to keep this issue open, so gonna close it.
(please feel free to re-open if you disagree)

Thanks again!

closed 2023-07-02 by lread

lread, 2023-07-16:
After some newbie soldering and resoldering, I'm here to report success with an OSH Park built PCB v2022-04-15 with a Teensy 4.1.
 
Thanks again for a fun and useful project, you have revived my dead Kinesis Advantage2!

stapelberg, 2023-07-17:
> I also had some newbiness with the molex connectors. I wasn't always inserting the ribbon deeply enough into the connector before closing. Do you want me to PR a tip to the README for this under troubleshooting?

sure

dm17, 2024-04-10:
The "full BOM" excludes the kinT and Teensy, correct? So we need only buy the "full BOM" and the correct Teensy - and we can then print the latest KinT PCB?