kinx-project/kint #72: Teensy 4.1 without ethernet compatible?

isaacgale, 2023-08-15:
I'm assuming the version without ethernet would be compatible with QMK but I can't find a definitive answer. Can anyone please confirm? It would also save me a few bucks in components too.

aleb, 2023-08-15:
If cost is a concern, it also works with Teensy LC.

isaacgale, 2023-08-15:
Thanks for the reply @aleb I'm set on a 4.1 (want leds too)  also the not ethernet models are more available at present where I am at least

aleb, 2023-08-15:
You can have three leds controllable with Teensy LC. Teensy 4.1 would allow to control all four. What else can you do with it and not with LC?

If "4.1 without ethernet" is the same as "4.1", then there should be no reason for it to not work. No ethernet functionality is used. Can you check if there are any incompatibilities between the two versions?

stapelberg, 2023-08-20:
This is the first time I hear about the “no ethernet” variant :)

Yes, that variant should work perfectly fine (though I haven’t tried it), as we don’t use any ethernet related functionality.

Please update this issue once you have it working.

isaacgale, 2023-11-29:
@stapelberg I can confirm that the non-ethernet 4.1 works.