kinx-project/kint #73: Ghosting with KB500 & Teensy 4.0

mil, 2023-08-24:
Hi @stapelberg  - thanks for your excellent work on the kinT replacement controller project.

I'm facing an issue where with the stock firmware (hex) I get ghosting on certain keys. The pattern is mainly the 2nd row column keys produce the key above:
- `w` produces 2 sometimes
- `e` produces 3 sometimes
- `r` produces 4 sometimes
- `p` produces 0 sometimes
- `o` produces 9 sometimes

I'm trying to figure out what could possibly be wrong. It seems really strange as sometimes the keyboard works normally, and then sometimes ghosting will occur with the keys above (and I can't seem to isolate the issue). I lost the original stock controller so I can't be 100% sure its a KinT issue; however playing with some QMK variables like `DEBOUNCE` etc does produce different results (though nothing 'correct')... so I'm not sure really the best variable to be playing with.

Also I thought maybe it had something to do with `delay_inline`, but that didn't help much either:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/kinesis/kint41/kint41.c#L55

Any help on where or how I could debug would be helpful, again the stock firmware .hex on github does produce the issue too, so maybe it's something specific to the matrix wiring on my KB500.

stapelberg, 2023-08-25:
My guess would be that this is wear & tear on either the key switches themselves (unlikely, increasing debounce time should help for individual key switches), the kinesis keywell PCBs, or the wiring between the PCBs and the micro controller.

Looking at the keyboard matrix, it seems like maybe COL_0 and/or COL_1 don’t have proper contact. Or maybe it’s the rows, not the columns? Easy to get confused, best to check both. You could try using the continuity test on your multimeter. Maybe that already shows something. Or inspect the PCB(s) for scratches. Or just try to re-solder the affected soldering joints proactively.

I hope any of this helps in your quest to find the problem!

mil, 2023-09-06:
Thanks for the feedback.

Turned out being an issue with soldering contact between the teensy and the PCB. Fixed it after all