kinx-project/kint #74: Request for JSON key config and Key Code Information

hellmind, 2023-09-14:
Hello,

I'm interested in obtaining information regarding the the JSON file containing the key mappings. My primary goal is to identify the key code used in the QMK firmware to trigger programming mode. Currently, when I use the key combination to enter programming mode ("qk_boot"), the device restarts, and this process takes a considerable amount of time. I would like to know if there is a more efficient way to enter programming mode and if there is a way to make the device exit this mode or restart automatically without having to reconnect the USB cable.

Additionally, I would appreciate guidance on the correct procedure for adding a new key to the device. If it is not defined in the matrix or if it is necessary to use keys that are on the keyboard's diagram, I would like to know how to proceed. If it's not possible to add a new key directly, I'm also considering the option of connecting it in parallel to another key, such as "keypd," but this may require soldering a diode to the Teensy 4.1 contacts, since im gonna use a pedal (simple switch, piano/keyboard sustain).

I am currently using three Teensy 4.1 devices. Any information or guidance you can provide would be greatly appreciated.

Thank you in advance for your assistance!

stapelberg, 2023-09-17:
This sounds like a general QMK question, so it would be better to ask the question in the QMK support channels.

I’m afraid I don’t know the answers to your questions, but maybe someone else can chime in.

hellmind, 2023-09-18:
Mr. Stapelberg, I wanted to express my sincere gratitude for all the hard work you've put into this project. It's truly an honor to receive your response and guidance.

I've made significant progress with the project of installing a pedal, but I've encountered some issues related to radio frequency interference (RFI). It appears that extending the keyboard matrix without any filtering or protection might not have been the best approach. When I transmit and the pedal is engaged, the keyboard goes haywire and sends multiple keypress signals.

I'm now exploring ways to effectively filter out this interference to ensure the smooth operation of the system. Your expertise and insights have been invaluable in this endeavor, and I appreciate your continued support.

As a radio enthusiast, my primary intention for implementing the pedal is to enhance my radio communication setup, specifically for push-to-talk functionality. This pedal will play a crucial role in facilitating seamless communication during my radio operations.

Thank you once again for your dedication and assistance.

aleb, 2023-09-19:
The issue you describe might be caused by "contact bouncing". Not sure what would be best in your case, a hardware debouncer or a software debouncing algorithm. See https://github.com/qmk/qmk_firmware/blob/master/docs/feature_debounce_type.md for what you can do with QMK.

hellmind, 2023-09-19:
I heard on some forums that the issue could potentially be resolved by
adding a 100-ohm resistor in series.
I'll try that first since it works perfectly when I'm not transmitting.