kinx-project/kint #75: red light blinking and cant flash. Keyboard doesnt type.

ussher, 2023-09-25:
[https://www.youtube.com/shorts/t0Yrm5OCTzw](url)

I've been using the current QMK firmware on the keyboard for about a week, its been working fine.  After dinner I went to use it and no key press had any output.  I tried unplugging and re-plugging, tried different usb ports and even tried switching the cable but its all the same, no keypress has any output.

I wanted to try re-flashing QMK .hex file but where pressing the button on the Teensy 4.1 used to cause an orange led to flash then the flashing would finish, now pressing that same button does not result in the orange led flashing.

The issue seams to be with the red flashing LED but im not sure what that means or steps to take to get the keyboad working again.

I bought the unit pre-assembled from a seller on ebay with teensy 4.1 installed.

This isn't really a problem with this code, but I've run out of other ideas on how to get it working.  Hopefully someone can point me in the right direction. thanks.

ussher, 2023-09-25:
It looks to me like the "2 Blinks = NXP JTAG Not Responding" code from here:
https://www.pjrc.com/store/ic_mkl02_t4.html

"....2 Blinks = NXP JTAG Not Responding
    No communication is working! The 2 likely causes for this problem are improper power startup sequence, or a problem with any of these connections between the MKL02 and IMXRT chip:......."

Not sure what to do about that though.

stapelberg, 2023-09-25:
Sounds like either a bad Teensy or a bad soldering contact. 

You could try re-soldering any soldering joints that don’t look 100%, or even just all joints.

If re-soldering doesn’t help, you probably need to put in a new Teensy. It looks like your build does not use a socket, so effectively you’ll need to buy a new board.

ussher, 2023-09-25:
argh, thanks.  I bought that one through ebay fully assembled.  I can get teeny from amazon here (japan) and do have a soldering iron but very little faith in my skills.   Since it did work for a while then stopped working my suspicion is your first idea 'a bad teensy' is the most likely.

I'll procrastinate a while on it.

Thanks for the help.

stapelberg, 2023-09-25:
> I can get teeny from amazon here (japan) and do have a soldering iron but very little faith in my skills.

Just to be clear: de-soldering a broken teensy and soldering in a new teensy is virtually impossible.

For my first tip of going over the soldering joints, you don’t need any new parts — just a soldering iron and the same kind of solder (lead vs. lead-free) that was used with the board. If you don’t trust your soldering skills, maybe there is a repair cafe, hackerspace or makerspace in your city or area with people who could help out.

But yeah, just buying a new fully-assembled board will be the easiest solution for sure.

closed 2023-09-25 by stapelberg