kinx-project/kint #76: JSON/keymap changes in latest QMK for kint41 et al

afonsoguerra, 2023-10-14:
Hi all,

Maybe I'm missing something obvious, but I just tried importing my old, tried and tested keymap JSON files to the qmk config website and it's now all over the place. I then changed the old keymap.c file I had by hand and reflashed the keyboard (also with the latest qmk) and ... it didn't go well. After a bit of github digging, it seems that the keymap layout for these keyboards changed recently (https://github.com/qmk/qmk_configurator/pull/1312). 

Does anyone have knowledge or a suggestion on how to port a JSON or c keymap (I know both are interchangeable) to the new layout arrangement? I would rather not have to reconfigure my 5 layers by hand if I can avoid it. 

Thanks for any help!

stapelberg, 2023-10-15:
I’m not sure — I haven’t really followed QMK development closely over the last few months.

In general, you will have the best chances for a good reply if you post this question at the QMK support channels.

afonsoguerra, 2023-11-25:
This turned into a case of scratching my own itch... but in the hope that it will help someone else in the same situation, where one has a complex, multi-layer keymap that they want to port to the latest version of QMK, I'm attaching the correspondence between the old and new layout. The file has three columns, position in old layout, position in new layout, key code from default keymap. 

[keymap_conversion.txt](https://github.com/kinx-project/kint/files/13464067/keymap_conversion.txt)

I'm happy to develop this further, submit a PR with a little script or whatever else is useful, if there is interest. 

Thanks again for all your work on this project, it's been a lifesaver for my hands in more ways than one.

mgsloan, 2024-01-06:
Thanks so much for the script @afonsoguerra !  Nice time saver even for my single layer layout.

conceptual-perception, 2024-02-25:
@afonsoguerra, I'm interested. The above issue on the QMK_Configurator project is mine.

I tried to get this working with my JSON.
[ariadne-4-jan-2023-v2.json](https://github.com/kinx-project/kint/files/14398191/ariadne-4-jan-2023-v2.1.json)

I get this error.

```
ariadne@tesseract:~/kinesis-kint$ python3 keymap_json_converter.py -i ariadne-4-jan-2023-v2 -o ariadne-4-jan-2023-v3

{0: 0, 1: 1, 2: 2, 3: 3, 4: 4, 5: 5, 6: 6, 7: 7, 8: 8, 9: 18, 10: 19, 11: 20, 12: 21, 13: 22, 14: 23, 15: 30, 16: 31, 17: 32, 18: 33, 19: 34, 20: 35, 21: 42, 22: 43, 23: 44, 24: 45, 25: 46, 26: 47, 27: 54, 28: 55, 29: 56, 30: 57, 31: 58, 32: 59, 33: 66, 34: 67, 35: 68, 36: 69, 37: 74, 38: 75, 39: 78, 40: 80, 41: 81, 42: 82, 43: 9, 44: 10, 45: 11, 46: 12, 47: 13, 48: 14, 49: 15, 50: 16, 51: 17, 52: 24, 53: 25, 54: 26, 55: 27, 56: 28, 57: 29, 58: 36, 59: 37, 60: 38, 61: 39, 62: 40, 63: 41, 64: 48, 65: 49, 66: 50, 67: 51, 68: 52, 69: 53, 70: 60, 71: 61, 72: 62, 73: 63, 74: 64, 75: 65, 76: 70, 77: 71, 78: 72, 79: 73, 80: 76, 81: 77, 82: 79, 83: 83, 84: 84, 85: 85}

The number of keys in the layer is different from the number of keys in the conversion dictionary, please double check the input files.
```

All the defined layers have 86 entries, the undefined layers just are `[],`

My JSON hasn't been modified.

### QMK Kinesis Layer Theory
| Layer | Kind          | Purpose                                               |
|-------|------------|----------------------------------------------------------|
| 0     | Always Active                    | Dvorak                             |
| 3     | Toggles to Active & Default      | QWERTY                             |
| 12    | Momentary                        | Number Pad                         |
| 13    | Momentary                        | Arrows                             |
| 15    | Momentary                        | Layer Toggles, Keyboard Management |

afonsoguerra, 2024-02-26:
Hey @conceptual-perception, I’ll be very happy to help you out, but I’m all over the place for a few days. Maybe next week (early March?). In the meantime, can you figure out why you have 86 keys and I seem to have 85? I’m sure you can just add another entry to the dictionary in the script, to make it 86, and it will run but I am not sure the effect will be a good one because I don’t know where the extra key is coming from.

afonsoguerra, 2024-02-26:
P.S. - The quick fix would just paste 86 “KC_TRANS”, in the empty layers in your original JSON :)