kinx-project/kint #77: My hex runs slow, like 10 seconds to boot.

hellmind, 2023-11-01:
I tried to use https://config.qmk.fm/#/kinesis/kint41/LAYOUT to create and compile my own layout.
It works but not as the default hex. 
the default hex layout takes no time to boot, mine  take like 10 seconds.
Whats wrong
Also i tried with :
git clone https://github.com/qmk/qmk_firmware.git
Then compiled the default  but the size isnt the same, and when I flash it, the boot times also went to 15s.
What can be wrong?

Maybe i should compile my own layout with a custom qmk version ?

aleb, 2023-11-02:
The default one you used was compiled by you?

stapelberg, 2023-11-02:
The default version linked from https://github.com/kinx-project/kint are pretty old builds at this point. 

You can try building an equally old QMK, then upgrade to a newer QMK to see if the problem is with the firmware itself or with the configuration.

hellmind, 2023-11-02:
> The default one you used was compiled by you?

I tried both, the default compiled by me, was like 3 times the size of https://github.com/kinx-project/kint/blob/main/default-firmware/kinesis_kint41_default.hex and boots slow.

Also I compiled it using my own linux and the configurator site withe same results.
The only  firmware that boots correctly is: https://github.com/kinx-project/kint/blob/main/default-firmware/kinesis_kint41_default.hex

> The default version linked from https://github.com/kinx-project/kint are pretty old builds at this point.
> 
> You can try building an equally old QMK, then upgrade to a newer QMK to see if the problem is with the firmware itself or with the configuration.

How can I do that?

stapelberg, 2023-11-02:
> I tried both, the default compiled by me, was like 3 times the size of https://github.com/kinx-project/kint/blob/main/default-firmware/kinesis_kint41_default.hex and boots slow.

Okay, so it sounds like the problem isn’t the configuration, but the software version then.

> > You can try building an equally old QMK, then upgrade to a newer QMK to see if the problem is with the firmware itself or with the configuration.
> 
> How can I do that?

I would recommend using git to check out an older version of QMK. You can try `git checkout 0.22.13` to check out version 0.22.13, for example. Try narrowing down the problem to a version that works well and the first following version that introduces the problematic behavior.

hellmind, 2023-11-04:
I found that the problem appears with any version starting from 0.22. Version 0.21.7 works fine.
I still need to compile my custom layout that I created with the QMK Configurator page. Using version 0.21.7, I'm not sure how to use the downloaded JSON file.

stapelberg, 2023-11-04:
> I found that the problem appears with any version starting from 0.22. Version 0.21.7 works fine.

Okay, great that we now have a version range.

If you want to go one step further, you can use “git bisect” to find the exact git commit between these two versions that introduced the problem. This will go a long way to getting the QMK developers to take the problem seriously.

hellmind, 2023-11-04:
This is the result:

`qmk_firmware (173f54c) (BISECTING) [!?] via C v13.2.1-gcc via 🐍 v3.11.5 took 5s ➜ git bisect good
416af0171c6433a7ecb198386dd2c3ac70d4cbd2 is the first bad commit
commit 416af0171c6433a7ecb198386dd2c3ac70d4cbd2
Author: Nick Brassel <nick@tzarc.org>
Date:   Sat Jun 24 05:31:25 2023 +1000

    Remove CORTEX_ENABLE_WFI_IDLE from keyboards. (#21353)

    * Remove CORTEX_ENABLE_WFI_IDLE from the codebase.

    * Turn on CORTEX_ENABLE_WFI_IDLE by default.

 keyboards/4pplet/aekiso60/rev_b/rules.mk              | 2 --
 keyboards/4pplet/eagle_viper_rep/rev_a/rules.mk       | 2 --
 keyboards/4pplet/eagle_viper_rep/rev_b/rules.mk       | 2 --
 keyboards/4pplet/perk60_iso/rev_a/rules.mk            | 2 --
 keyboards/4pplet/steezy60/rev_b/rules.mk              | 2 --
 keyboards/4pplet/unextended_std/rev_a/rules.mk        | 2 --
 keyboards/4pplet/waffling60/rev_d/rules.mk            | 2 --
 keyboards/4pplet/waffling60/rev_d_ansi/rules.mk       | 2 --
 keyboards/4pplet/waffling60/rev_d_iso/rules.mk        | 2 --
 keyboards/4pplet/waffling80/rev_b/rules.mk            | 2 --
 keyboards/4pplet/yakiimo/rev_a/rules.mk               | 2 --
 keyboards/acheron/apollo/87h/delta/rules.mk           | 2 --
 keyboards/acheron/apollo/87h/gamma/rules.mk           | 2 --
 keyboards/acheron/apollo/87htsc/rules.mk              | 2 --
 keyboards/acheron/apollo/88htsc/rules.mk              | 2 --
 keyboards/acheron/arctic/rules.mk                     | 2 --
 keyboards/acheron/athena/alpha/rules.mk               | 2 --
 keyboards/acheron/athena/beta/rules.mk                | 2 --
 keyboards/acheron/austin/rules.mk                     | 2 --
 keyboards/acheron/elongate/delta/rules.mk             | 2 --
 keyboards/acheron/keebspcb/rules.mk                   | 2 --
 keyboards/acheron/lasgweloth/rules.mk                 | 2 --
 keyboards/acheron/shark/beta/rules.mk                 | 2 --
 keyboards/aeboards/ext65/rev2/rules.mk                | 2 --
 keyboards/ai03/andromeda/rules.mk                     | 2 --
 keyboards/ai03/vega/rules.mk                          | 2 --
 keyboards/akegata_denki/device_one/rules.mk           | 2 --
 keyboards/alas/rules.mk                               | 5 -----
 keyboards/alpine65/rules.mk                           | 2 --
 keyboards/at_at/660m/rules.mk                         | 2 --
 keyboards/atreus/f103/rules.mk                        | 2 --
 keyboards/aurora65/rules.mk                           | 5 -----
 keyboards/baion_808/rules.mk                          | 5 -----
 keyboards/binepad/bnr1/v2/rules.mk                    | 2 --
 keyboards/boston/rules.mk                             | 2 --
 keyboards/box75/rules.mk                              | 2 --
 keyboards/bt66tech/bt66tech60/rules.mk                | 2 --
 keyboards/cablecardesigns/phoenix/rules.mk            | 4 ----
 keyboards/cannonkeys/aella/rules.mk                   | 2 --
 keyboards/cannonkeys/an_c/rules.mk                    | 2 --
 keyboards/cannonkeys/bakeneko60_iso_hs/rules.mk       | 2 --
 keyboards/cannonkeys/bakeneko65_iso_hs/rules.mk       | 2 --
 keyboards/cannonkeys/balance/rules.mk                 | 2 --
 keyboards/cannonkeys/brutalv2_65/rules.mk             | 2 --
 keyboards/cannonkeys/chimera65/rules.mk               | 2 --
 keyboards/cannonkeys/chimera65_hs/rules.mk            | 2 --
 keyboards/cannonkeys/cloudline/rules.mk               | 2 --
 keyboards/cannonkeys/crin/rules.mk                    | 2 --
 keyboards/cannonkeys/db60/rules.mk                    | 2 --
 keyboards/cannonkeys/devastatingtkl/rules.mk          | 2 --
 keyboards/cannonkeys/gentoo/rules.mk                  | 2 --
 keyboards/cannonkeys/gentoo_hs/rules.mk               | 2 --
 keyboards/cannonkeys/instant60/rules.mk               | 2 --
 keyboards/cannonkeys/instant65/rules.mk               | 2 --
 keyboards/cannonkeys/iron165/rules.mk                 | 2 --
 keyboards/cannonkeys/leviatan/rules.mk                | 2 --
[... 245 lines cut ...]
 keyboards/xelus/rs60/rev2_1/rules.mk                  | 2 --
 keyboards/xelus/trinityxttkl/rules.mk                 | 2 --
 keyboards/xelus/valor_frl_tkl/rev1/rules.mk           | 2 --
 keyboards/xelus/valor_frl_tkl/rev2_0/rules.mk         | 3 ---
 keyboards/xelus/valor_frl_tkl/rev2_1/rules.mk         | 3 ---
 keyboards/xelus/xs108/rules.mk                        | 2 --
 keyboards/ymdk/id75/rules.mk                          | 2 --
 keyboards/zoo/wampus/rules.mk                         | 2 --
 platforms/chibios/config.h                            | 7 +++++++
 platforms/chibios/platform.mk                         | 9 +++++++++
 311 files changed, 29 insertions(+), 755 deletions(-)
 mode change 100755 => 100644 keyboards/acheron/elongate/delta/rules.mk
 mode change 100755 => 100644 keyboards/alas/rules.mk
 mode change 100755 => 100644 keyboards/baion_808/rules.mk
 mode change 100755 => 100644 keyboards/binepad/bnr1/v2/rules.mk
 mode change 100755 => 100644 keyboards/cablecardesigns/phoenix/rules.mk
 mode change 100755 => 100644 keyboards/hand88/rules.mk
 mode change 100755 => 100644 keyboards/keychron/q11/ansi_encoder/rules.mk
 mode change 100755 => 100644 keyboards/mnk1800s/rules.mk
 mode change 100755 => 100644 keyboards/mnk50/rules.mk
 mode change 100755 => 100644 keyboards/mnk75/rules.mk
 mode change 100755 => 100644 keyboards/novelkeys/nk65b/rules.mk
 create mode 100644 platforms/chibios/config.h`

Also, how can I compile my QMK Configurator keymap.json with an older version? I tried converting it to keymap.c and downloading the entire QMK Configurator project to obtain that keymap.c, but neither method worked.

hellmind, 2023-11-04:
Commenting

#ifndef CORTEX_ENABLE_WFI_IDLE
#    define CORTEX_ENABLE_WFI_IDLE TRUE
#endif // CORTEX_ENABLE_WFI_IDLE

from platforms/chibios/chibios_config.h
fixed my problem

isaacgale, 2023-12-01:
I've noticed a similar thing. I'm using 21.6 if I pull latest from master compile and flash my keymap it seems to brick my keyboard (it becomes unresponsive) until I re-flash it with my older complied keymap. I'm using a teensy 4.1 without ethernet. 

I wanted to go to latest so I could use Caps Word with Key overrides. But ended up just cherry-picking that commit to my 21.6 based branch for now.

isaacgale, 2023-12-03:
It looks like you should be able to set `OPT_DEFS += -DCORTEX_ENABLE_WFI_IDLE=FALSE` in your rules.mk file instead of commenting out the above. I was going to test this out but I see the layout has also changed now according to 'kinesis: remove stacked split layouts (#21569)' 18251eb8da78485507a58c177c3635da88b935cb

I don't currently have the time to rework my layout right now.

stevenbetten, 2024-02-11:
I created https://github.com/qmk/qmk_firmware/issues/23053 and https://github.com/qmk/qmk_firmware/pull/23054 to address this for kint41. I don't have any pre-4.1 teensy controllers so I cannot test them and thus did not make any changes to kintlc, kint35, or stapelberg.