kinx-project/kint #78: Teensy 4.1 with headers

edmonds, 2024-01-02:
Hi,

I saw the answer to https://github.com/kinx-project/kint/issues/43 and watched the soldering video, but I wanted to confirm. If I don't need any LEDs, can I avoid soldering those components (the SMD LEDs, resistors, and the vertical pin header), and use the [Teensy 4.1 DEV-20360](https://www.sparkfun.com/products/20360) variant that includes pre-attached headers? Or are those components required for some reason?

Thanks!

stapelberg, 2024-01-03:
The LEDs, their resistors and the vertical pin header can be left off; this is even listed as an advantage of the kinT design:

> The kinT makes possible lower-cost builds: if you don’t need the scroll lock, num lock and keypad LEDs, you can use a Teensy LC for merely 11 USD.

Do note that if you buy a Teensy with headers, you might need to clip off the headers below the board in case they stick out too much (not entirely sure if that’s a problem).

closed 2024-01-03 by stapelberg

edmonds, 2024-02-03:
Just to close the loop on this, I built a kinT without LEDs using the Teensy 4.1 with pre-attached headers and it worked just fine. I trimmed the headers down to a proper length for soldering, not sure if it would have fit into the case without trimming but they do stick out a fair bit if not trimmed.

[image]