kinx-project/kint #79: Kint Controller mapped to RP2040 - Possible not-too-tedious solution

oneleaftea, 2024-02-26:
Sorry if this Issues section is not the best place to talk about this. No issue with the Kint controller, but I have been looking into STM32 or RP2040 solutions that can be dropped into the Kint controller with some modification. I have a bunch of PCB's so would prefer to use them rather than make a new board based on a new controller (I know the Black Pill version is out there as well, but I've had more experience with the RP2040 and love how easy it is to load the firmware).

I believe I have finally found a solution. Some background:

- I have built with the Waveshare RP2040-Zero and the classic super-cheap RP2040 Pico. The latter would be the best option, with 29 GPIO pins. However, the biggest problem is that the across-board pin-to-pin width of the board is 2.54mm wider than the Teensy (17.78mm for the RP vs 15.24mm for the Teensy).
- Most of the Pro-Micro form factor RP2040 have too few pins.
- But I finally found the Keebio Elite-Pi, which has the same form factor as the Pro-Micro (so same across-board pin-to-pin width as the Teensy) but has the extra pins at the bottom edge. It has 25 GPIO pins (GP10 and GP11 are connected to ground unless two traces are cut, which their documentation covers). This allows us to connect 3 LED's. The below picture would leave off the Keypad LED, but you could pick any LED to leave off.

In the below picture, the black rectangles show the header pins you would use, where you can connect directly to the Kint controller PCB. So basically 12 headers on one side and 8 headers on the other. Plenty stable IMO. The rest would have the board hanging and wires going from the original Teensy-based holes to the corresponding Elite-Pi pins. As you can see, it would take 7 wires.

Based on my picture, I think this is a fairly easy procedure. I've done much worse. Any thoughts or see any possible issues? Appreciate any thoughts! 

[image]

stapelberg, 2024-02-26:
Thanks for opening this issue! 

I don’t have time to look at the wiring details right now (sorry!), but I just wanted to say that this is a good place to talk about a RP2040 version of the board :)

oneleaftea, 2024-02-26:
Great! And just a quick note to anyone following. I found another cheaper board (it's a Pro Micro footprint RP2040 on Aliexpress with 27 GPIO pins. So it would cover all the LED's we need with a pin to spare. But until I get it in my hands, I'm not as confident in it as I am with the Elite-Pi. It would also need 10 wires, so a bit more tedious than the Elite-C. I did make a wiring diagram, but will doublecheck it a bit before I post it.

oneleaftea, 2024-03-08:
For those that want more detailed instructions and a proper wiring diagram, you can see this repo I just setup: https://github.com/oneleaftea/KinT_RP2040_Mod