kinx-project/kint #80: Teensy Loader on 4.1 reboots into HalfKay (requires unplugging and replugging keyboard)

oneleaftea, 2024-03-01:
Has anyone had this issue with Teensy 4.1? When using the Teensy Loader, I am able to program and reboot, but it goes right back into HalfKay. The verbose output shows that it is still online and ready to be flashed. And so the keyboard doesn't work unless I unplug and replug in the keyboard. The firmware is properly loaded and no error in the verbose output (it does say something about the .elf file and it being an unknown board though?).

Is the only way to get out of programming mode to unplug and replug in the keyboard? 

I have also noticed that the Teensy Loader seems to take a minute after pressing the reset button before it is ready to flash. 

Curious what other people's experiences are with flashing the unit? Otherwise, it is working great and all the features I need are working. Thanks again for such a great project!

oneleaftea, 2024-03-01:
Oh, I just found out that the Issue #77 actually could be related to this. I just implemented the fix that @isaacgale proposed to my rules.mk, and the bootloader went perfectly smooth this time! Rebooted directly into a working keyboard without having to unplug and replug.

Feel free to close or we can leave this open for visibility. Thanks!

stapelberg, 2024-03-04:
Would you mind sending that fix upstream to QMK? Thank you

oneleaftea, 2024-03-04:
> Would you mind sending that fix upstream to QMK? Thank you

Hello, I does seem like someone attempted a pull request on this and it was rejected. [Pull Request # 23054](https://github.com/qmk/qmk_firmware/pull/23054).

I read over the pull request and felt like the submitter made as good a case as any for them to accept the pull request, but they concluded that it is a fix that needs to be taken up with ChibiOS.