kinx-project/kint #81: Jumper locations and other teensy++ clarifications?

256k, 2024-03-27:
Hello, I've have the boards printed for a long time and i've finally come around to installing it into my kinesis.

I just had a few notes and questions to clarify to be sure i dont make a mistake:

1. the teensy++ you say i need to cut traces:
"Using a small knife such as a hobby knife, cut the traces between the pads of jumpers JP4, JP5, and JP6. This will disconnect pin 7, pin 15 and pin 16." 
I'm not sure exactly where those jumpers are? on the kinT board i only see J1/J2/J3 next to it the text: "Compat w/ teensy++"
has the board design changed but not the documentation? 

2. I'm modifiying a very very old kinesis (The essential: KB132PC) with the hard PCB molex connector points so i will be handwiring the connectors into the board. anything about this super old board that i would need to know? from my limited research it seems like the wiring should be the same?

3. looking at the schematics pdf i see there's 2 references to wiring:  
https://github.com/kinx-project/kint/blob/44e6c8be96a0e1e13ada5eafdeba8c51a2d6c9e8/schematic-v2020-06-23.pdf
"original matrix wiring" and "HumbleHacker matrix wiring"
I'm not sure i understand the difference? is this something i need to take into consideration?

Thank you for this great work and giving new life to older kinesis boards that might be a bit too long in the tooth.

stapelberg, 2024-03-27:
Can you share which revision of the kinT you have? That will allow me to give a more helpful answer.

256k, 2024-03-27:
yes of course sorry for not clarifying. looking at the board seems i have `v2020-06-30`

stapelberg, 2024-03-28:
You can browse the changes since v2020-06-30 here: https://github.com/kinx-project/kint/compare/v2020-06-30...main

In particular, this is the README as of the version you have: https://github.com/kinx-project/kint/tree/ae11c62d8600c0019aa810abf71808ad528f5c99?tab=readme-ov-file — here you’ll find the soldering instructions that match your version of the board.

See https://github.com/kinx-project/kint/issues/42 for more details on older versions like the Kinesis Essential.

You don’t need to care about the matrix wiring, I think. The kinT controller uses the HumbleHacker matrix wiring, which was an older hobbyist replacement controller from 10+ years ago. This is only documented so that the full schematic is available in case you need to tinker more or adapt the board to other keyboards, or similar. 

Hope this helps

256k, 2024-03-28:
fantastic thank you so much. 

one last question, would you be able to tell me the part number/specs of the ribbon cable that would connect 2 of those molex connectors together? since i have the older kinesis it's a pain in the a** to wire it manually and i was hopingto use a flexible ribbon like on the newer kinesis instead. just dont know exactly what is the part number i need to get

stapelberg, 2024-04-09:
Unfortunately I don’t have that part number handy, as I never bought a spare FFC ribbon cable.

What I can say:

Look for a molex 13-pin FPC/FFC cable with 1.25mm spacing.
It should be compatible with https://octopart.com/39-53-2135-molex-7670149?r=sp in case that helps.

Please share the part number you ended up with in the end :)