kinx-project/kint #83: Left keywell row intermittent issues

ttttcrngyblflpp, 2024-07-05:
The left half of my keyboard has intermittent issues where sometimes a row of keys, e.g. the home row, will not work, or pressing a key in a row causes all other keys in the row to be actuated as well. When this happens I usually open up the keyboard and put it back together and then it goes away again...

I've tried following the troubleshooting resources available, which makes me think that this isn't an issue with the PCB, but possibly with the soldering on the keywell itself, probably the ribbon cable port? Any help would be appreciated.

stapelberg, 2024-07-06:
> I've tried following the troubleshooting resources available, 

Can you list what specifically you have tried? Maybe something else comes to mind then.

> which makes me think that this isn't an issue with the PCB, but possibly with the soldering on the keywell itself, probably the ribbon cable port? Any help would be appreciated.

Could be anything. Most common are bad solder points. You could just re-solder all connections relevant to the affected keywell.

Less common, but also possible are bad traces on the PCB (caused by wear and tear). In that case, you would need to solder in a cable to bridge the sometimes-broken trace. These are hard to find, though — maybe apply a voltmeter in continuity testing mode and flex the PCB a bit to see if the contact stays or is flaky. 

Hope this helps and good luck with finding the issue!

afonsoguerra, 2024-07-06:
Just to add my experience. With use and if you take the molex ribbons in and out a few times, they might get bent corners and as such they can go in slightly sideways. This causes some tracks to not connect properly. I suggest you remove the molex ribbons from both sides, clean it a bit, inspect it and re-seat it properly and see if the problem goes away. You can also swap ribbons within the keyboard and see if the problem shifts to a different area of the keyboard. Thankfully, the ribbons themselves are not too hard to source and replace if that is your problem.

stapelberg, 2024-07-07:
@afonsoguerra Could you share where you sourced a working ribbon cable? Do you have a link / model name perhaps? Thank you

afonsoguerra, 2024-07-07:
I got mine from RS: https://uk.rs-online.com/web/c/?searchTerm=molex+ribbon&sra=oss&r=t

I'll make a note to check for the part-number and I'll edit this when I find the order.

stapelberg, 2024-07-08:
Thanks @afonsoguerra, I added this info to the README :)

ttttcrngyblflpp, 2025-01-13:
I ended up ordering new PCBs and soldering a whole new set of parts together from leftover parts and it seems like it's working. As a software engineer who is a novice to soldering I did not have any respect for the possibility of soldering issues, so let this be a warning to anyone reading this that make sure you don't have any cold joints or not-enough-solder joints.

closed 2025-01-13 by ttttcrngyblflpp