kinx-project/kint #88: Boot key does not set Teensy 4.1 to bootloader mode in custom firmware

sockbot, 2025-09-09:
When I use the default firmware on a Teensy 4.1, the mapped boot key sets the Teensy into bootloader mode.

When I compile a custom firmware with the configurator at https://config.qmk.fm/#/kinesis/kint41/LAYOUT pressing the boot key makes the Teensy and the keyboard unresponsive until power cycled.

To reproduce, compile this keyboard layout and press the boot key:

[kinesis_kint41_layout_2025-09-08.json](https://github.com/user-attachments/files/22224271/kinesis_kint41_layout_2025-09-08.json)

sockbot, 2025-09-09:
Symptoms may be related to this issue https://github.com/kinx-project/kint/issues/77

stapelberg, 2025-09-09:
Is this a duplicate of issue #80?

sockbot, 2025-09-13:
I'm not sure how to duplicate #80. I'm using teensy_loader_cli -- how do I see the state of the Teensy? I'm just going to try the fix in #77 to use an older version of qmk as a workaround.

sockbot, 2025-10-21:
I was able to work around this issue by checking out the `issue-23053` branch here https://github.com/stevenbetten/qmk_firmware/tree/issue-23053 and using it to compile the firmware.

I note the qmk documentation failed to install all of the dependencies on Arch so some manual work is required, but it worked in the end.

closed 2025-10-21 by sockbot

stapelberg, 2025-10-21:
Let’s keep this issue open for visibility until it’s fixed in QMK upstream.